slic3r/Slic3r #4867: Vase mode generates glitchy infill where it shouldn't

douira, 2019-09-27:
**Describe the bug**
When slicing in vase mode some layers have solid parts generated near the edges although there should be no infill there. (even if vase mode was off, no infill should be generated there)

**To Reproduce**
Try to slice the provided model with the provided settings.

**3D Model and Slic3r Configuration Export**
[Archive.zip with config, vase model and generated gcode](https://github.com/slic3r/Slic3r/files/3664223/Archive.zip)

**Expected behavior**
Expected all layers above the bottom solid layers to not have any solid infill.

**Screenshots**
[image]
[image]
[image]

**Desktop**
 - OS: macOS 1.14.6
 - Version 1.3.0

**Additional context**
This also happens on 1.2.9 apparently. I could not get the 1.3.1 dev version to generate gcode at all. (the preview was just blank) Also, enabling or disabling vase mode doesn't seem to change anything, the setting might also just be broken. Judging from the gcode it seems as if the vase mode setting has no effect at all. But still, the infill here is glitchy even for vase mode off.

supermerill, 2019-10-10:
Seems like it create bottom layers "not on baseplate". You can try to create a box modifier that set the bottom layers to 0 for the whole object but the first layers.

douira, 2019-10-10:
ok so doing that makes it work for me. But it doesn't really solve the problem since it still generates bad gcode when you use vase mode without this. I suggest either doing this automatically when vase mode is enabled or adding the requirement for such a modifier to the list of settings that need to be applied in order to activate vase mode when the box is checked.

EvilGremlin, 2019-10-19:
Oh, that's because of current vase logic. Implementation of #4114 will fix that, i'm going to dig into that matter.