slic3r/Slic3r #4868: Different infill angle for each layer

frghirardini, 2019-10-03:
I am trying to print with a carbon fiber reinforced nylon with a 100% rectilinear infill. I would like to set a different infill angle for each layer instead of having it set by the default 90° (cross-hatching). Is there a way of doing it?

lordofhyphens, 2019-10-04:
Not without code modifications

frghirardini, 2019-10-04:
Yes I meant with code modifications

supermerill, 2019-10-14:
you have to modify _layer_angle() in Fill.hpp or the overrided version in specialized fill classes.

malv-c, 2019-10-21:
hi
and what about the angle calculated with perimeters to be infilled
or relative angle not the angle done once for the printer
it should be a very powerfull infill for full pieces ;)

shreyakrishna, 2020-11-19:
Can someone post the code to modify the layer angle