slic3r/Slic3r #4871: Unrequested bridging

krhackbarth, 2019-10-15:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
A clear and concise description of what the bug is.
I'm getting bridging over a hole in the model but I haven't requested it.

**To Reproduce**
Steps to reproduce the behavior:
1. drag and drop STL file
2. click on Preview tab

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.
see attached:
[Files.zip](https://github.com/slic3r/Slic3r/files/3727422/Files.zip)

**Expected behavior**
A clear and concise description of what you expected to happen.
there shouldn't be any bridging over the hole

**Screenshots**
If applicable, add screenshots to help explain your problem.
see zip file
**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
Windows 10 Pro
 - Version [e.g. 1.3.0]
1.3.0
     - _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

**Additional context**
Add any other context about the problem here.
Problem doesn't occur in PrusaSlicer.

foreachthing, 2019-10-15:
I don't see this in version 1.3.1:

[image]

krhackbarth, 2019-10-17:
Not sure if I did this right but I'm still seeing it in my dev version of 1.3.1 -- dated 2019-09-23 23:17.
[image]
[image]

foreachthing, 2019-10-17:
You've done nothing wrong - I just had an older version installed. I downloaded the latest build but get the same result: Everything looks just fine!
[image]

Are you sure you use the same model as the one you attached?
Try one of the attached models; I've run it through [Blender](https://www.blender.org/) with the "3D-Printing" add-on and the other through [netfabb online service](https://service.netfabb.com/service.php).
[fixed-Keyguard_V22.zip](https://github.com/slic3r/Slic3r/files/3737419/fixed-Keyguard_V22.zip)

krhackbarth, 2019-10-21:
Interesting, the Blender version shows the bridging both in 1.3.0 and in 1.3.1.  The netfabb version shows no bridging in either version.  As I mentioned earlier in the thread, PrusaSlicer shows no bridging.  I wonder if it is doing some post-processing of the STL... The STL itself is generated by OpenSCAD.

krhackbarth, 2019-10-21:
FWIW, I have supports turned on from the build surface and no supports for bridging.

krhackbarth, 2019-10-21:
I can also get rid of the bridging by running the original STL through 3D Builder (built into Windows 10).

trysten, 2021-08-06:
Could not replicate on 1.3.0 or 473319b92