slic3r/Slic3r #4873: Slic3r DOESN NOT WORK on OSX Catalina

CPISolutionsAV, 2019-10-17:
Alright, well I am sad to say I had to update my mac last night because of a problem I was having. I was excited to see that Catalina fixed this problem, but I was very sad to see that after the update, several apps didn't work anymore. One of these apps is Slic3r. From what I have read, it is because its not optimized for 64-bit and Apple has finally blocked apps running if they are not. 

So this leaves me with either running slicer on a VM windows machine, or getting a cheap Windows laptop just to run Slicr.

What are the chances to Slic3r getting an update to make it compatible? I would think within a year that would mean there would be hardly any Mac users using Slic3r anymore.

I dont want to, but I am going to start trying to figure out how to use Cura with my FFCRP but I did not have luck in the past.

Thanks.

PifPof73, 2019-10-18:
Hi,
same problem for me.
Does anybody know, if we can build slic3r ourselfs from the source?
Are there any constraints (libs, APIs etc.) which makes compiling for Mac OSX 64bit difficult?

I will try to follow these instructions and see, if we have to adjust it for 64 bit compatibility:
[https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-OS-X](url)

@CPISolutionsAV - you should have some patience, many software projects are just adapting this issue, often it is not much effort to produce a 64bit build.

I will post a comment if I have any further information / success.
Regards,
PifPof73 (a huge fanboy of this great peace of software)

lordofhyphens, 2019-10-18:
"not optimized for 64 bit" is news to me.

I don't have a Catalina system myself (and I barely use my Mojave mini), so
testing is going to be "fun".

Have you tried a build from:
 https://dl.slic3r.org/dev/mac/jenkins/

PifPof73, 2019-10-19:
HI,
the tried [https://dl.slic3r.org/dev/mac/jenkins/slic3r-1.3.1-dev-2ef957da-osx.dmg](url), which is running on my Mac OSX with Catalina.
Thanks @lordofhyphens for the link.
Issue can be closed.
Nice weekend to all,
PifPof73

xyzdevops, 2019-12-03:
@CPISolutionsAV run the linux version in a docker container

lordofhyphens, 2019-12-06:
@CPISolutionsAV can you check https://mega.nz/#!8aoxhCbR!L-OMNmFp8K7h1og-1fzLAIn2DO3sPFui3TPN8Z8Dlvg ?

lordofhyphens, 2019-12-06:
I'm mostly interested in knowing if it's just Travis being garbage (that was built on my local system) or if it's something in the current packaging scripts.

guzforster, 2020-06-16:
For me the error is that "developer cannot be verified". Tried the latest build from Jenkins, same error. Tried the proposed one on the link, same error. Any tips?

lordofhyphens, 2020-06-17:
I probably need to go in and update the developer certificate. You can tell
OSX to ignore the issue in the Security options last I recall.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2020-06-17:
@alranel IIRC it's your cert specifically, can you update the certificate used on both systems used for signing?

tahoematt, 2020-08-04:
I can't install on Catalina also. No options in security to allow like there is sometimes.