slic3r/Slic3r #4874: Update Install Docs For Redhat Family of Distros

holmanb, 2019-10-21:
**Is your feature request related to a problem? Please describe.**
1. Installing Slic3r on RHEL/CentOS is undocumented.
2. The install documentation on Fedora is dated and incomplete.

**Describe the solution you'd like**
I would like to create a pull request to improve the documentation for RHEL and Fedora.

holmanb, 2019-10-21:
I see that the docs are part of the wiki, thus a pull request is not required.  I've added an update for Fedora in the wiki, and will do the same for RHEL after I get slic3r working on RHEL.

dwillmore, 2019-11-04:
Have you had any luck with this?  I believe I was the last one to update the Fedora entry and that was probably over a year ago.  If you have completed your updates, please feel free to close this issue.  Thank you for your contribution.

holmanb, 2019-11-04:
Thanks for checking @dwillmore.  The Fedora updates to the wiki are complete, but I'm still struggling through getting a clean RHEL install. It looks like avoiding system perl is the way to go on RHEL, but I can only work a little bit at a time on this, so the going is slow.  My current recommendation to future users would be to avoid RHEL7.  Unfortunately for me, that is outside of my control.  I will definitely update the wiki when I get a reproducible process, or just close the ticket when I give up.

jamitch-snl, 2019-12-20:
Hi Brett,

I just started trying to get a build on RHEL 7.  I have made some progress but still having issues.  I think I have it mostly compiling but the build system is complicated (I don't know perl), and the approach to the testing and install is foreign to me.  

I'm just trying to build the command line tools and the underlying library.

Like me, it sound like you are tied to RHEL 7.  

Perhaps if we put our heads together we can get this to work?  

I will be back in office at start of new year.  
John

lordofhyphens, 2019-12-21:
Gcc 4.8.5 iirc doesn't support enough c++11 and 14 to build Slic3r

lordofhyphens, 2019-12-21:
I have had to learn how to do things on rhel 7 for work reasons. Still not a fan

holmanb, 2019-12-26:
@jamitch-snl 

I'll be back around the same time.  I'm happy to collaborate and share in greater detail the progress I've made.

jamitch-snl, 2020-01-07:
Brett, and/or anyone else who may be able to help.

I would like to continue discussion on building slic3r on RHEL 7 systems.  

For the moment, I am just trying to build command line tool without gui.

RHEL 7 system is highly incompatible with slic3r requirements.  Here is what I have done so far.

- installed perlbrew
- installed cmake 3.16.2
- installed GCC 9.2

I run the build (this doesn't use cmake)

- perl Build.PL

Some progress; have a compile without errors --managed to point to correct boost headers but its linking with wrong version of boost libraries.  Basically I hacked xs/Build.PL to point to correct headers but haven't managed to get it to link with correct boost libraries.  There are multiple boost installations on my system.

Shared library is created:
xs/blib/arch/auto/Slic3r/XS/XS.so

But before installation, build system attempts running tests, some of which pass, and some fail.  Test failures seem to cause the overall installation to fail.

??? Questions ???

- Is there a way I can manually run tests to debug? -- before install?  What tests would I run?  How do I run them?
- Is there a way to skip running of the tests?
- Any suggestions on hacking xs/Build.PL regarding boost libraries;  There is a specific directory /lib64/boost169/ with needed libraries but I haven't yet figured out how to get it to use those rather than a different version installed in the common area /usr/lib64/; I don't want to play around with symbolic links on the system.  I just want to get the build to point at the correct boost headers and libraries.
- The library link line doesn't have -L/<directory to boost libs>; When I run ldd on XS.so it points to wrong boost libraries.
-  I'm not exactly sure what is going to be installed?  Is it a series of perl packages/modules? Or is it just the one command line tool slic3r.pl?
- What is the finally structure of installation?  System looks like a series of perl scripts/module/packages that link with C++ code -- is that correct?
- What does xs (XS) stand for?

jamitch-snl, 2020-01-07:
@lordofhyphens: you mentioned working RHEL 7.  What did you have to do to get a build of slic3r?  Do you have any notes you can share?

lordofhyphens, 2020-01-07:
I don't currently have an rhel7 build.

jamitch-snl, 2020-01-09:
Update: Managed to get slic3r built (without gui) on rhel 7.  Here is a summary of what I did.  I moved away from using perl build approach; instead I used cmake.

**I did install perlbrew and compatible perl; I'm not sure if this has much of an effect on the build using cmake.**

- installed perlbrew and perl: perl5/perlbrew/perls/threaded-perl-5.24.4/bin/perl
- installed GCC 9.2
- installed cmake 3.16.2
- installed boost_1.72.0

**I used GCC 9.2 to compile cmake and boost.**  Note that boost has a configuration file that is very helpful to cmake; otherwise you have to specify  path to boost include and libs just right for it to find them correctly.

- boost_1.72.0/lib/cmake/Boost-1.72.0/BoostConfig.cmake 

**I also had to tweak CMakeLists.txt file to not use static libraries for boost.**

- SLIC3R_STATIC=OFF

I ran test:

- slic3r_test -s

test cases:  50 |  47 passed | 3 failed as expected
assertions: 718 | 715 passed | 3 failed as expected

I was also able generate gcode from an stl file using a binary for slic3r which I downloaded.  Similarly I was able to use binary that I built on the stl file.

lordofhyphens, 2020-01-10:
CMake build does not need Perl at all, and builds a different CLI