slic3r/Slic3r #4875: update VCTargetsPath.vcxproj

malv-c, 2019-10-22:
to compile git with m$c++ 2019 give :
    La génération a démarré 22/10/2019 12:43:13.
    Projet "E:\git\slic3r\build\CMakeFiles\3.16.0-rc1\VCTargetsPath.vcxproj" sur le noud 1 (cibles par défaut).
    E:\Program Files (x86)\Microsoft Visual Studio\2019\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targets(775,5): error : The OutputPath property is not set for project 'VCTargetsPath.vcxproj'.  Please check to make sure that you have specified a valid combination of Configuration and Platform for this project.  Configuration='Debug'  Platform='Visual Studio 16 2019 Win64'.  You may be seeing this message because you are trying to build a project without a solution file, and have specified a non-default Configuration or Platform that doesn't exist for this project. [E:\git\slic3r\build\CMakeFiles\3.16.0-rc1\VCTargetsPath.vcxproj]
    Génération du projet "E:\git\slic3r\build\CMakeFiles\3.16.0-rc1\VCTargetsPath.vcxproj" terminée (cibles par défaut) -- ÉCHEC
first time m$c++ here idk how to fix sorry

lordofhyphens, 2019-10-22:
I do not build with Visual Studio. This has probably something to do with settings in the CMake filen and suggest a Google search along those lines.

malv-c, 2019-10-23:
ok i will try
thanks