slic3r/Slic3r #4877: If-statements in "before layer change g-code" box won't get parsed

tehdusk, 2019-10-24:
**Describe the bug**
I cannot get any if-statement in "before layer change g-code" to be parsed correctly.
Note even `{if [layer_num] == 10}M600` as stated in the manual for conditional g-code seems to work for me.
Scripts like this for Temp-towers
```
{if layer_z==1.6} M104 S225
{elsif layer_z==11.6} M104 S220
(...)
{endif}
```
fail the same way. The Elsifs seem to work though, only any if-statements are ignored (and cut). Also see screenshot:
[image]
Even something like {if 1==1} does not work.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to printer settings -> custom g-code
2. enter any if-statement into the "before layer change G-Code box", for example `{if [layer_num] == 10}M600`
3. Export the G-code and open it with a texteditor like notepad or notepad++.
Instead of `{if [layer_num] == 10}M600`, only ` M600` is found in the G-code.
At the end, the correct statement can be found though as a comment:
`; before_layer_gcode = {if [layer_num] == 10}M600`

3D-model, config and GCode:
[config+3DModel+GCode.zip](https://github.com/slic3r/Slic3r/files/3769712/config%2B3DModel%2BGCode.zip)

**Expected behavior**
I expected the code to be parsed successfully. This does not happen with example codes from the manuals nor with example scripts which seem to work for other users, as the gcode does not contain the wanted custom code.

**Desktop (please complete the following information):**
 - OS: Windows 10 x64 1903
 - Version 1.3.0 and 1.3.1dev 4f5b935ecf1335755d607ec38395bd94106daa7f fail the same way for me.

VanessaE, 2019-10-25:
As far as I know, the `{if}{elsif}{endif}` syntax is only valid in the Prusa3D fork, or anyway it never worked for me. Here on upstream, you need to do something a bit different.  I am also at commit 4f5b935ecf1335755d607ec38395bd94106daa7f, and the following is the code that I use (I just now double checked that it still works):

```
; temperature tower:
; (10 mm segment height, -5°C per segment, starting as described below,
; with 1.5 mm thick base below the first segment)
; (SmartTemperatureTower_xxxx.stl)

; starting at 225° (PLA)...
; {if [layer_z] > 1.5 }M104 S{ 225 - floor( { {[layer_z] - 1.5} / 10} )*5}

; starting at 250° (PETG, ABS)...
; {if [layer_z] > 1.5 }M104 S{ 250 - floor( { {[layer_z] - 1.5} / 10} )*5}

; insert custom commands at specific layer heights:
;
;{if {
;	{{[layer_z] > 5.99} and {[layer_z] < 6.01}}
;	or
;	{{[layer_z] > 33.99} and {[layer_z] < 34.01}}
;}}M600
;
;{if {
;	{{[layer_z] > 6.99} and {[layer_z] < 7.01}}
;	or
;	{{[layer_z] > 34.99} and {[layer_z] < 35.01}}
;}}another G-code here

```
Uncomment the section(s) of code you need.  Note that with the temperature tower code, I did not bother to limit the temperature commands to appear only on the exact layers they're needed on, but the changes themselves do occur at the right points (so while every layer above 1.5 mm gets a command, the temperatures are all the same within any given tier of the model).

tehdusk, 2019-10-25:
I'll try that when I'm back home. But the script aside, the example `{if [layer_num] == 10}M600` from the manual should be working, correct?

VanessaE, 2019-10-25:
I should think so, but only if there's no code preceding that trips-up the interpreter.

tehdusk, 2019-10-25:
So I just tried with the example from the manual again. As before, it does not work.
I also tried it with a much simpler model. Same thing here. I also tried putting it into the "after layer change g-code" box, same result.
Zip of config & resulting gCode attached.
[Uploading otherModel+Manual-Example__config+GCode.zip…]()

I then tried two lines from your code. This also doesn't get parsed.
[image]
[image]

tehdusk, 2019-10-25:
So to solve this, there's some behaviour from the slicing I did not anticipate, and in addition to that, I got confused on how the resulting line should look like.

The main problem however was that the script I was given, the layer number just didn't exist, and so the IF was not true.
For example, where I expected layer_z==91.55, the actual value was 91.5500000000008, and that resulted in false, not parsing the code. After defining ranges instead, it's working as anticipated.

Thanks for your help!

closed 2019-10-25 by tehdusk

Oekn5w, 2019-10-25:
Hmm, all in all I think we should aim towards what you initially wrote with `if .. else if /elif / elsif .. endif` maybe with an keyword to account for floating point inaccuracies.

StanSPP, 2019-11-10:
I had the same two issues attempting a Temperature Tower in Slic3r 1.3.0.   First the 'if elsif' logic available in the PE branches is not in the base Silc3r code so you need to modified the gcode to work with a series of separate 'if' statements (see my final code below).  Secondly, I was also getting the odd errors in expected layer_z values.  All my layer heights are set to 0.2 but for some reason Slic3r is producing minuscule errors (e.g. 31.5999999999999 and not the expected 31.6), therefore my workaround was to add the upper and lower bounds for each section of the tower.

----Modified and tested code----
;BEFORE_LAYER_CHANGE, Z = [layer_z]
{if [layer_z]>=1.6}{if [layer_z]<11.6}M104 S225
{if [layer_z]>=11.6}{if [layer_z]<21.6}M104 S220
{if [layer_z]>=21.6}{if [layer_z]<31.6}M104 S215
{if [layer_z]>=31.6}{if [layer_z]<41.6}M104 S210
{if [layer_z]>=41.6}{if [layer_z]<51.6}M104 S205
{if [layer_z]>=51.6}{if [layer_z]<61.6}M104 S200
{if [layer_z]>=61.6}{if [layer_z]<71.6}M104 S195
{if [layer_z]>=71.6}{if [layer_z]<81.6}M104 S190
{if [layer_z]>=81.6}{if [layer_z]<91.6}M104 S185
{if [layer_z]>=91.6}M104 S180

kdkwarhead, 2019-12-28:
Thanks StanSPP!
Your code worked perfectly in Slic3R 1.3.0 (this was driving me creasy)

zbrozek, 2020-07-11:
I'd certainly like to understand why the code in the manual fails to parse though. This is verbatim pasted from the manual and it doesn't parse:
```!!!!! Failed to process the custom G-code template before_layer_gcode
Parsing error at line 5. Expecting a boolean expression.
{if [layer_num] == 10}M600
    ^
!!!!! End of an error report for the custom G-code template before_layer_gcode```