slic3r/Slic3r #4881: does not take into account the time of the layer on the outer perimeter, there is no slowdown

ShohninDmitriy, 2019-11-04:
Build commit: 2ef957dab6fbd65d7e00e5b6fd61cdf80670b565 version 1.3.1-dev

the time of the layer on the outer perimeter is not taken into account, there is no deceleration, although all the rest (inner layer, filling) decreases the speed

dwillmore, 2019-11-04:
Could you better describe the problem?  Slic3r tries to make sure that a layer takes at least as long as the minimum layer time.  It does that by slowing down perimeters, infill, bridges, and solid areas so that the total time meets this limit.  It *does not* slow down external perimeters because that could cause artifacts in the print.  But, it should consider the time it takes to draw the external perimeter in the layer time calculation.

Are you saying it is not taking into account the external perimeter time when performing the layer time calculation?  Or are you saying that the external perimeters are not slowed down?  If the latter, then this is the expected behavior.

supermerill, 2019-11-04:
It' s intended. if you reduce the speed, the external perimeter will have a different look and this will be visible.

closed 2019-11-05 by ShohninDmitriy