slic3r/Slic3r #4883: Tiny errors in [layer_z] values

StanSPP, 2019-11-10:
**Description of Bug**
While trying to create conditional gcode for a temperature tower (https://www.thingiverse.com/thing:2729076) I noticed that the output for [layer_z] contained tiny errors which caused problems when I was testing for exact values.  For example, I was attempting to change the temperature at layer height of 31.6 with "{if [layer_z]==31.6}M104 S210" but I couldn't find the temperature change statement in the outputted gcode file.  I discovered this is because the actual value reported was 31.5999999999999, not 31.6.  Therefore the conditional gcode is working correctly but the logic was not firing because [layer_z] was never exactly 31.6.

What is even more puzzling is the pattern of the errors.  The first one I can see is [layer_z] = 23.3999999999999, not 23.4 as expected. It remains slightly under until it recovers to the correct value at 34.6 until I see 41.6000000000001 where I'm expecting 41.6.  The error seems to grow gradually in .0000000000001 increments to 0.0000000000009 until a sudden drop back to .000000000001 when I expect layer-z to be 100.

I managed to work around this issue by setting ranges for [layer_z] rather than absolute values, e.g. "{if [layer_z]>=31.6}{if [layer_z]<41.6}M104 S210".  However, I think the layer_z values should to be fixed to stop others having similar issues in the future.

**To Reproduce**
Steps to reproduce the behavior:
1. Set 'Layer Height' and 'First Layer Height' to 0.2 in Print Settings
2. Add ';BEFORE_LAYER_CHANGE, Z = [layer_z]' to 'Before layer change G-code' section in Printer Settings>Custom G-code.
3. Load tower model https://www.thingiverse.com/thing:2729076 and export gcode
4. Examine gcode file for reported Z layer height value from step 2 command.

[SmartTemperatureTower_PLA_180-225.zip](https://github.com/slic3r/Slic3r/files/3828419/SmartTemperatureTower_PLA_180-225.zip)

**3D Model and Slic3r Configuration Export**
[config.zip](https://github.com/slic3r/Slic3r/files/3828423/config.zip)

**Expected behavior**
All [layer_z] values to be exact multiples of 0.2

**Screenshots**
Not applicable.

**Desktop (please complete the following information):**
 - OS: Windows 7 Ultimate, SP1
 - Version: 1.3.0

**Additional context**
Add any other context about the problem here.

lordofhyphens, 2019-11-10:
Floating point math errors.

VanessaE, 2019-11-10:
It seems to me that since Slic3r can display the exact layer height/position in the 3d preview (at the bottom of the slider), without floating precision glitches, it should be possible to do proper comparisons, too, if it were to apply the same rounding (or truncating) to `layer_z` et al. as is applied to the displayed number.

easytarget, 2019-12-09:
This was also hitting me; I use `layer_z` to generate text messages to be sent to a simple (serial/ascii) display. with a maximum of 10 chars on this the overflow was a problem.

However, at least for me, there is a reasonable workaround, I use a [Value Expression](https://manual.slic3r.org/advanced/conditional-gcode#value-expressions-g-code-math) and the `round()` function in [ExprTK](http://www.partow.net/programming/exprtk/index.html) to round the value down to 2 decimal places like this:

`{round([layer_z] * 100)/100}`

I Can then use this in my layer change gcode like so:
`M118 P2 S"{floor([layer_z] * 100)/100}mm"`

Which adds this to the generated gcode:
`M118 P2 S"39.45mm"`
Much better than the previous:
`M118 P2 S"39.4499999999998mm"`

This should be available anywhere conditional gcode works..

ledvinap, 2019-12-09:
Reducing number of significant digits (a bit below floating point resolution) when printing should fix this