slic3r/Slic3r #4884: What printerdevice ID do I enter with a creality CP-01? it wont accept the name

punkinctr, 2019-11-10:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

dwillmore, 2019-11-10:
Please fill out the template and provide an explanation of the problem.  The subject line with a vague description is insufficient.

punkinctr, 2019-11-10:
I am trying to install slicr3d and it asks for print device ID but does not
accept the name. What do they want. It’s a CrealityCP-01

dwillmore, 2019-11-10:
Can you explain what you mean by "does not accept the name"?

dwillmore, 2019-11-10:
OS may be important.  Version of Slic3r certainly is...

punkinctr, 2019-11-16:
when it asks for printer device ID and I put "creality CP-01" it doesn't
recognize the name  I would like to know what to enter there as the program
will not open without that filled in

dwillmore, 2019-11-16:
I can't reproduce this.  Can you try taking out the '-'?

punkinctr, 2019-11-16:
> when it asks for printer device ID and I put Creallity CP-01it doesn't

I have now found a copy of the SD card that came with the printer however
when i try to download from it I get an error message that reads- Installer
integrity check failed Common causes include incomplete download & damaged
media contact installers author to obtain new copy
more info at http;/NSIS.sf.net/NSIS.Error
Please can you send me the information that is on this SD card.
I am sorry for all the time of yours I'm using and do appreciate your help
Bil

punkinctr, 2019-11-16:
further to last email what i described happens when I try to open creality
workshop -en2019 9-10 exe
And I still don't know what to fill in where it asks for Printer Device ID

lordofhyphens, 2019-11-16:
Um, creality workshop? This is Slic3r, we cannot support creality workshop.

punkinctr, 2019-11-16:
Sorry wrong email