slic3r/Slic3r #4886: Specify separate default extrusion width for odd and even layers to improve strength.

ksgeek, 2019-11-11:
I would like to see a way specify different extrusion width for even/odd layers to improve strength.  Think of a brick laying patters are overlapped.  Anything to stagger the extrusion edges by layer.  
current way

pp|xxxxx|xxxxx|xxxxx
pp|xxxxx|xxxxx|xxxxx
pp|xxxxx|xxxxx|xxxxx

This can be achieved in a couple ways.  Simple even/odd parameter with multiple extrusion width specified.  But keep the perimeter extrusion the same for visual consistency.  

pp|xxxxx|xxxxx|xxxxx 250%
pp|xxxx|xxxx|xxxx 200%
pp|xxxxx|xxxxx|xxxxx 250%

A second way would be to keep the default extrusion width the same but give an even or odd layer perimeter extrustion value.  This would have the benefit of keeping the other shell layers at max extrusion with and speeding up the print.   

pp|xxxxx|xxxxx|xxxxx 250%
pppp|xxxxx|xxxxx|xxxxx 250% but alter the perimeter.  
pp|xxxxx|xxxxx|xxxxx 250%

foreachthing, 2019-11-11:
I like the "Cura"-way: alternate wall thickness (an extra perimeter).

supermerill, 2019-11-12:
here an inmplementation:
https://github.com/supermerill/Slic3r/commit/004ec5480d38ec7fb3a456e70fb63e0c3892b9d6