slic3r/Slic3r #4888: Slic3r won't open

rafaelguima2, 2019-11-13:
I always used slic3r, it was a few weeks that I did not use, when I tried to open it today I could not, just frozen on this screen, already tried to download a newer version and happened the same thing, I use WIN10 64-bit.
http://prntscr.com/pwkhy4

lordofhyphens, 2019-11-14:
Looks like graphics drivers.
In the  future, prefer attaching images directly to github.

rafaelguima2, 2019-11-14:
I updated the driver to the latest version and it's still the same, it's a rtx 2080

supermerill, 2019-11-14:
> to the latest version

Maybe it will work better with an old one?

rafaelguima2, 2019-11-14:
I was actually already using an old driver, but I installed the newer one to see if it worked again, I installed an old driver again and it still didn't work

lordofhyphens, 2019-11-15:
Is there something you can use to test OpenGL on your card?

rafaelguima2, 2019-11-15:
I ran the furmark and these were the results:
http://www.ozone3d.net/gpudb/score.php?which=823863
if you have any better way to test this please tell me, I also tested with this site shown in the print

[image]

rafaelguima2, 2019-11-19:
man this sucks, I really like slic3r, at this time it could do anything that I need, I literally don't know what else to do

foreachthing, 2019-11-19:
Try [Prusa](https://github.com/prusa3d/PrusaSlicer/releases); it's forked from Slic3r. Maybe this will work...

rafaelguima2, 2019-11-20:
Just downloaded prusaslicer and it works normally, I will try to replicate my original slic3r settings and see if it works well

lordofhyphens, 2019-11-20:
Part of the problem is reproducibility. If I cannot get it to break on my system I am hamstrung in trying to fix it

foreachthing, 2019-11-20:
What if you run `Slic3r-debug-console.exe`? _Maybe_ you'll get some output there?!

rafaelguima2, 2019-11-20:
[image]
Just happens the same thing

incognit024, 2019-12-02:
I am having same issue, very annoying that i cant get slic3r to open. Cura, Prusa slic3r works fine but i dont want to use those, i want slic3r.. tried all tips from other forum posts..

incognit024, 2019-12-03:
i gave up, just using prusa slic3r now

lordofhyphens, 2019-12-04:
It has something to do with the 3d/GL packaging interfacing with your local drivers. 

The builds from dl.slic3r.org/dev/win start fine on my win10 64 bit system.

incognit024, 2019-12-06:
thx for the reply, its funny that prusaslicer just stopped working right now also.. im looking at the files and confused on which one to grab

incognit024, 2019-12-06:
it still doesnt work

foreachthing, 2019-12-06:
_Maybe_ try the [STUDIO](https://www.nvidia.com/Download/driverResults.aspx/154342/en-us) drivers (check supported devices and OS as well).
I got an RTX 2070, Nvidia Version 441.28 and all is well ...

lordofhyphens, 2019-12-06:
@incognit024 interesting. That works on my Mac mini 2012 (it was built and packaged there). 

Even the package works. I will probably create a local user on it to check

incognit024, 2019-12-06:
ok so now repetier host is not opening.. its doing the same thing that slicer was doing.. its opening the window but never fully loads and gets stuck at " not responding"..

lordofhyphens, 2019-12-06:
Oh, derp. Forgot this one is Windows not osx (where there is something going on and I have been trying to sort it out)

rafaelguima2, 2019-12-06:
hey @incognit024, did you played RDR2? 
I was thinking and realized that it stopped working after installing RDR2, it installed some things from Vulkan, maybe this have any relation?

rafaelguima2, 2019-12-06:
> it still doesnt work.. i feel like it has something to do with it opening on my other monitor.. im stuck with cura now.. i have 2070 gpu with updated drivers.. what other drivers should i check?

I turned off and unplugged one of my monitors and guess what ...
Worked perfectly
maybe @lordofhyphens can reproduce this now?

lordofhyphens, 2019-12-06:
.... My dev system for Windows only used 1 monitor.

incognit024, 2019-12-14:
repetier host is doing the same thing now.. when i open it, it opens on my second monitor and it doesnt respond

incognit024, 2019-12-17:
I got slic3r working with the "2372 " file "2373" didnt work for me https://dl.slic3r.org/dev/win/