slic3r/Slic3r #4891: Slic3r 1.3.0 crashing on STL add

Undeadlord, 2019-11-20:
I am running the latest production version, 1.3.0 x64. 

**Describe the bug**
I seeing a full crash when attempting to open a specific STL file. The file opens fine in Cura 4.3 and any export I can do from there, to 3MF or OBJ also crashes Slic3r in the same fashion. I have attached the STL.

From my event log,
Faulting application name: Slic3r.exe, version: 1.3.0.0, time stamp: 0x5af22f9b
Faulting module name: ntdll.dll, version: 10.0.17763.831, time stamp: 0x1f1a0210
Exception code: 0xc000041d

**To Reproduce**
Steps to reproduce the behavior:
1. Open Slic3r and click on the Add button.
2. Select the mentioned STL, 3MF or OBJ file.
3. Slic3r crashes.

**3D Model and Slic3r Configuration Export**
[Stirge.zip](https://github.com/slic3r/Slic3r/files/3872054/Stirge.zip)

**Expected behavior**
Expecting the STL to open

**Desktop (please complete the following information):**
 - OS: Win10 1809 x64
 - Version 1.3.0

foreachthing, 2019-11-21:
Your model is garbage. Sorry! ;-)
I've run it through [netfabb](https://service.netfabb.com/service.php) and all is fine.
When opening your model in [Blender](https://blender.org) and let "3D Printing" check it, there are several problems: 253 non manifold edges, 107 intersecting faces .... 
### Solution
**fix it through netfabb/Blender.**

My FEM-Software highlights faces with wrong normals in pink:
[image]
Pink: Elements facing the wrong way.

And here is the model's problem (or at least one):
[image]
There are loose faces at the base.

foreachthing, 2020-07-16:
No responds. Closing.

closed 2020-07-16 by foreachthing

PetNoire, 2020-08-07:
having the program hard crash for any reason, bad file or not, is still a bug