slic3r/Slic3r #4893: Bottom View (Ctrl + 5 ) is not as expected.

DocTrucker, 2019-11-25:
**Describe the bug**
The bottom view isn't working as expected.

**To Reproduce**
Load box.stl

Ctrl+4 for "Top" View.
View is ok.

Ctrl+5 for "Bottom" View.
View is NOT ok.

Ctrl+6 for "Left" View.
View is ok.

Ctrl+7 for "Right" View.
View is ok.

Ctrl+8 for "Front" View.
View is ok.

Ctrl+9 for "Back" View.
View is ok.

Ctrl+0 for "Diagonal" View.
View is ok.

**Expected behavior**
No views other than diagonal should show any more than a coloured square face and all other faces hidden.

**Screenshots**
See screeenshots.

**Desktop (please complete the following information):**
Slic3r Version: 1.3.1-dev
Slic3r Build Commit: 4f5b935
Ubuntu 18.04.

**Additional context**
N/A

[191125_Slic3r_Bug2.zip](https://github.com/slic3r/Slic3r/files/3886884/191125_Slic3r_Bug2.zip)

supermerill, 2019-11-25:
You see the axis.
it's intended.

DocTrucker, 2019-11-25:
You can see two faces of the part indicating it is at an angle to the x/y plane. It isn't!

All other views are ok.

supermerill, 2019-11-25:
your ctrl5 screenshot is a bit difficult to analyse, as it's very zoomed in.
The darker green area on top indicate that it's not showing against the (0,0,1) vector but at an angle?
i can't see an entire bed square, so it's not possible to analyse from that.

DocTrucker, 2019-11-25:
I was just grabbing screenshots as returned from ctrl 4-9 & 0 as the zoom Slic3R defaults to to avoid and ambiguity. Essentially a cube with all faces parallel to xy, xz, or yz planes should only show one face in any of the Ctrl4-9 views.

I'm adding the screenshots back into the original post.