slic3r/Slic3r #4894: Cut feature ignores no auto align preferences.

DocTrucker, 2019-11-25:
**Describe the bug**
Parts are moved after cutting despite all auto align preferences being disabled.

**To Reproduce**
"File" menu > "Preferences" > Deselect "Auto-centre parts (x,y):" & "Auto-align parts (z=0):"
(See screen shot & ini)
Click ok to clear popup.
Load benchy
Left click to select, right click to raise menu. > "Cut"
Keep upper and lower part, cut level at 24mm.
(Screenshot)
Click perform cut
(Screen shot)

Parts have been moved, rather than maintained original position.

**Expected behavior**
Cut parts retain original locations until explicitly moved or re-centred.

**Screenshots**
See screeenshots.

**Desktop (please complete the following information):**
Slic3r Version: 1.3.1-dev
Slic3r Build Commit: 4f5b935
Ubuntu 18.04.

**Additional context**
N/A

supermerill, 2019-11-25:
That's not as easy as it is. Cut is made to print a difficult piece that will be glue afterward.
Cutting is a bit like creating a new 3D model, and its center should be on the same spot.
How this new models (in fact, there are 2 model in 2 different parts) should be created is decorrelated from the "auto align" preference.

DocTrucker, 2019-11-25:
I don't follow your post there.

I'm looking to perform a cut and leave the parts in the original orientation. I have de-selected all auto align features, so don't expect anything to move after the cut. Moving the parts after the cut is afterall an additional operation?

supermerill, 2019-11-25:
a cut is an operation that destroy the mesh and create two new meshes. Theirs center isn't the center of the old one. It's impossible to "not move".

DocTrucker, 2019-11-25:
Are you sure?

Mesh data is usually relative to a specific point. Ok, you need to subdivide all triangles that intersect the defined plane and create new ones to fill the missing face, but in carrying out that operation the two halves would remain in the same place?

This doesn't appear to match what is happening in the screenshots either.

If there was some sort of movement of all the vertex datta to be relative to the same space the two meshes would lie in the same space, not have the top half neatly shifted down (-y axis move) a little bit and the neatly re-aligned to z=0?

supermerill, 2019-11-25:
can you do just a little test to prove me i'm wrong?

add 2 object in the platter
move the first object to the right side
move the second object to the left side
cut the second object
Does it re-arrange everything?

DocTrucker, 2019-11-25:
I will do exactly what you were asking but was halfway through doing something else first. I'm fairly sure it is doing a (normally helpful) move to avoid the two parts existing in the same place so...

Load box:
[image]

Cut Settings:
[image]

Cut Result
[image]

It may just be that it is doing an align on the two new parts rather than the whole bed. Ok, working on what you request...

DocTrucker, 2019-11-25:
Definately appears to be an uninvited "Align" operation creeping in. My second part was another box but measuring 50x50mm. Cut plane was z=10.

Pre-cut layout:
[image]

Cut window:
[image]

Result (top view):
[image]

Result (Diagonal view):
[image]

foreachthing, 2019-11-25:
If you want to split parts (i.e. for multi-color printing) you'll need to "add" to an existing part!
1. Perform cut
2. EXPORT model (i.e. TOP cut) as stl (or something else)
3. Double-click the BASE and `Load Part...`
4. Select and ok TOP part
5. Position TOP part in xyz (and maybe the BASE as well.
6. And then it should look like this:
[image]

Sliced:
[image]

DocTrucker, 2019-11-25:
Ok following on from a comment in another thread I can see that the object model is intended to allow parts to be aligned relative to each other and be moved about the build platform in one lump. Nice. However it doesn't slove what I perceive as a bug that prevents me using the cut feature in Slic3r.

In this next work I have created a FreeCAD file with two objects that together form the 10x10x10mm cube. Meshed and exported them in freeCAD, then load them and I get the following as intended (highlight on the lower part for clarity):

[image]

Following on the comments about the object model in Slic3r I can load the lower part, then go through the settings pop up and load the second. Now the two parts can be moved as one. Nice:

[image]

But now I try to cut the block as before export each half as STL, delete all, load one, then open the settings menu and try to load the second. The Cut operation has clearly destroyed the information relating tothe alignment of the two blocks.

[image]

It is appears to me that there is a 'convienience' align operation occurring here that according to my preference settings should not be occurring.

I suggest the following should be considered for fixing this issue:

Set the cut operation to respect the auto align (x,y) and z=0 settings from the preferences menu.
Add an additional option of the cut pop up window to create an object containing both halves in a similar way to how one is created if you add part through the settings pop up on another part rather than add part on the toolbar.