slic3r/Slic3r #4895: Part offset in z axis doesn't work.

DocTrucker, 2019-11-25:
**Describe the bug**
Part offset in z axis doesn't work.

**To Reproduce**
"File" menu > "Preferences" > Deselect "Auto-centre parts (x,y):" & "Auto-align parts (z=0):"
(See screen shot & ini)
Click ok to clear popup.
Load benchy
Left click to select, right click to raise menu. > "Cut"
Keep upper and lower part.
(Screenshot)
Click perform cut
(Screen shot)
Parts have been moved unexpectedly. (see other bug)
Left click select top part of model.
Right click to raise menu. > "Settings"
Click "#3DBency.stl" in tree window.
Type "24" into "Z:" text entry box in the "Move" section of the popup window.
(Screen shot)
Press ok.
Popup window goes, part hasn't moved as expected.
Ctrl + 8 for front view.
(Screen shot)
Top section still on z=0 plane.
Ctrl + 0 for diagonal view.
Left click select top part of model.
Right click to raise menu. > "Settings"
Click "#3DBency.stl" in tree window.
Type "24" into "Z:" text entry box in the "Move" section of the popup window.
Press enter key.
Top section still on z=0 plane.

**Expected behavior**
Top part should have been offset to z=24 level.

**Screenshots**
See screenshots.

**Desktop (please complete the following information):**
Slic3r Version: 1.3.1-dev
Slic3r Build Commit: 4f5b935
Ubuntu 18.04.

**Additional context**
N/A

[191125_Slic3r_Bug4.zip](https://github.com/slic3r/Slic3r/files/3887016/191125_Slic3r_Bug4.zip)

supermerill, 2019-11-25:
warning: i can't see the screenshot so maybe i'm wrong.

I think that cut create 2 pieces not 1 piece with 2 parts.
If you save the top part as stl, remove it from slic3r, then go to settings of the bottom part, then add part and select the saved stl, then you should be able to move it.

DocTrucker, 2019-11-25:
Screenshot should be in the zip file, let me check.

supermerill, 2019-11-25:
you can drag-drop them

DocTrucker, 2019-11-25:
I'll bear that in mind for future bug reports/feature requests. Just running through your suggestions.

DocTrucker, 2019-11-25:
Ok i have removed the previous work flow from cutting. try the following.

**Describe the bug**
Part offset in z axis doesn't work.

**To Reproduce**
"File" menu > "Preferences" > Deselect "Auto-centre parts (x,y):" & "Auto-align parts (z=0):"

[image]

Click ok to clear popup.
Press "Add" toolbar button, select box, press open.
Press "Add" toolbar button, select box, press open.
Left click to select a box.
Right click for pop-up menu and select settings.
Type "10" into "Z:" text entry box in the "Move" section of the popup window.

[image]

Press ok.
Ctrl + 8 for front view.

[image]

Both boxes still on z=0 plane.

**Expected behaviour**
Top part should have been offset to z=10 level.
Front view should show rectangular object.

**Screenshots**
Screenshots also in zip file.

**Desktop (please complete the following information):**
Slic3r Version: 1.3.1-dev
Slic3r Build Commit: 4f5b935
Ubuntu 18.04.

**Additional context**
N/A

[191125_Slic3r_bug4b.zip](https://github.com/slic3r/Slic3r/files/3887159/191125_Slic3r_bug4b.zip)

lordofhyphens, 2019-11-25:
Gotta offset relative to something else

DocTrucker, 2019-11-25:
That doesn't make sense to me. I want to offset relative to world/machine co-ordinates, can you explain or point to how to move a part to z=[a specific value] in the manual?

Why does the preview show it working ok, but then the result fail? What am I doing wrong?

DocTrucker, 2019-11-25:
Here's more context. I found this bug when trying to resolve the other bug about cutting shifting parts back down to z=0 despite all auto align features being disabled.

Model:
[image]

Cut window:
[image]

Cur result:
[image]

Manual xy alignment (bad, but alignement data has been lost):
[image]

Attempt to move upper part back up by 2.2 (not as bad as the 2.2 was explicitly specified before, but still extra unneeded workflow):
[image]

Result of no move, feet still below base level:
[image]

foreachthing, 2019-11-25:
If you want to split parts (i.e. for multi-color printing) you'll need to "add" to an existing part!
1. Perform cut
2. EXPORT model (i.e. TOP cut) as stl (or something else)
3. Double-click the BASE and `Load Part...`
4. Select and ok TOP part
5. Position TOP part in xyz (and maybe the BASE as well.
6. And then it should look like this:
[image]

Sliced:
[image]

DocTrucker, 2019-11-25:
Not multicoloured printing, but may as well be. Makes little odds.

Odd workflow. In that case shouldn't the controls to offset the part be hidden when it clearly can't be achieved with a single part?

foreachthing, 2019-11-26:
@DocTrucker, yes! I, too, find it odd but got used to.
I think it's a good thing, objects drop to the plate. If you miss something and your print starts 0.5 mm in the air ....
You could add a cube (make it really tiny) and add your object to the cube. Then z-offset it as desired.

DocTrucker, 2019-11-26:
Yeah the work around for offsetting is possible just more work than it needs to be for the users.

It's also not consistant. Load an STL that is floating (my two half box example) and it respects the don't align to z=0 option.

I would hope that I have shown that it would make sense for all opperations to