slic3r/Slic3r #4896: Preview not working

surdu, 2019-12-01:
**Describe the bug**
Preview doesn't work on MacOs for the latest build from master

**To Reproduce**
Steps to reproduce the behavior:
1. Download the latest MacOS build
2. Load any stl
3. Go to the Preview tab
4. See nothing
5. Same for Layers tab
6. Also 'Send to printer' doesn't seem to send anything, though the save gcode works

**Desktop (please complete the following information):**
 - OS: MacOS 10.14.6 (18G1012)
 - Version: 1.3.1-dev master (66af505)

lordofhyphens, 2019-12-01:
Try: https://dl.slic3r.org/dev/mac/jenkins/ 

There's something going on with the Mojave builds that I have not been able to sort out.

lordofhyphens, 2019-12-05:
@surdu re: send to printer, are the permissions set correctly for the serial device for your printer (if it's a usb connection)?

surdu, 2019-12-05:
@lordofhyphens It's an OctoPrint server, and it works on the release build and Jenkins build. I'll test it again after the preview issue is fixed. I feel that they are related. If it's still an issue afterwards, I'll create a separate ticket.

lordofhyphens, 2019-12-05:
Weird.

lordofhyphens, 2019-12-06:
@surdu can you try this build? https://mega.nz/#!1exm1QzT!AbFusHhb68JjwSoIn2ScpgF0dRswSXv7mQCDv8bcLgs 

You'll have to tell Gatekeeper that it's fine in order to run it (it's unsigned)

lordofhyphens, 2019-12-06:
I'll get my personal code signing sorted out tonight.

surdu, 2019-12-06:
@lordofhyphens Both preview and send to printer works 😉

lordofhyphens, 2019-12-06:
... well that is weird as someone else with a related problem is still having issues, but you aren't?

So that tells me at least part of it is building on Travis at all, and not how we are building on Travis.