slic3r/Slic3r #4898: New Icon proposal

stockHuman, 2019-12-02:
Hello good people,

I'd like to propose a fresh coat of paint on the app icon. See it right here:
<p align="center">[image]</p>

And here with context, next to blender and a custom icon for the fork of CNCjs by SnapMaker.

[image]

If the emerald green is too far a departure from the original icon, I could tone it down - though I find it works next to other icons and on its own. The rationale for the change was the icon not really fitting with the rest of my system's icons, and being somewhat hard read, visually speaking. 

Test it out for yourselves with [icns](https://github.com/slic3r/Slic3r/files/3912407/slic3r.icns.zip), <a href="https://github.com/slic3r/Slic3r/files/3912393/favicon.ico.zip" download>ico</a> or <a href="https://user-images.githubusercontent.com/13852458/69980202-c3a6ee80-14fd-11ea-99cd-b29dec443cb5.png" download>a large png</a>.

Or render it out on fellow [FOSS software](https://github.com/slic3r/Slic3r/files/3912403/slic3r.blend.zip).

Thanks!

foreachthing, 2019-12-02:
I like it "flat", like the Blender one. But I like the emerald colors a lot! :-)

stockHuman, 2019-12-03:
Thanks for your input! :) I tried flat designs on the original icon, but going from 'cartoon' to flat on a sliced sphere effectively makes it two half circles - and that's why I went with the rendered polyhedron. Anyway, here are some flat designs - with more and less pronounced offsets in the slices. I find the first one most effective of the flat ones.

[image]  [image] [image]

lordofhyphens, 2019-12-03:
@alranel opinions?

VanessaE, 2019-12-03:
I dunno about him, but in *my* opinion :smile: the proposed image (whichever of the four is desired) ought be supplied as a themed icon via `/usr/share/icons/FOO` for whatever theme `FOO` best matches the given style (or however that works on non-*nix platforms).

Slic3r's existing icon can then become the `hicolor` standard image, or whatever passes for the "fallback" icon if no themed image is provided.

Either that or make it a local user override.

(See also the discussion on IRC)

closed 2019-12-31 by stockHuman