slic3r/Slic3r #4899: Dev download site unavailable

fitzterra, 2019-12-03:
Hi all,

I'm getting a _Forbidden_ error when trying to get a dev build from http://dl.slic3r.org/dev/
I've only noticed it since last week Friday and thought it may be a temporary issue, but I see it is still the same this morning.

Are nightly builds still available (for Linux in my case)?

Apologies if this should not be reported here.

Thanks

foreachthing, 2019-12-03:
Works on my end (Win, Linux and Mac).

fitzterra, 2019-12-03:
Very strange. I get this from various net connections, and various machines - all Linux using Firefox and/or Chrome or the machines. I doubt it, but could it be with connecting from South Africa?

Here is output from testing with curl:

```
$ curl -vL https://dl.slic3r.org/dev/
*   Trying 92.243.6.202...
* TCP_NODELAY set
* Connected to dl.slic3r.org (92.243.6.202) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Client hello (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS change cipher, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES128-GCM-SHA256
* ALPN, server did not agree to a protocol
* Server certificate:
*  subject: CN=dl.slic3r.org
*  start date: Nov 26 22:53:34 2019 GMT
*  expire date: Feb 24 22:53:34 2020 GMT
*  subjectAltName: host "dl.slic3r.org" matched cert's "dl.slic3r.org"
*  issuer: C=US; O=Let's Encrypt; CN=Let's Encrypt Authority X3
*  SSL certificate verify ok.
> GET /dev/ HTTP/1.1
> Host: dl.slic3r.org
> User-Agent: curl/7.52.1
> Accept: */*
> 
< HTTP/1.1 403 Forbidden
< Date: Tue, 03 Dec 2019 09:50:48 GMT
< Server: Apache/2.4.7 (Ubuntu)
< Content-Length: 285
< Content-Type: text/html; charset=iso-8859-1
< 
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>403 Forbidden</title>
</head><body>
<h1>Forbidden</h1>
<p>You don't have permission to access /dev/
on this server.</p>
<hr>
<address>Apache/2.4.7 (Ubuntu) Server at dl.slic3r.org Port 443</address>
</body></html>
* Curl_http_done: called premature == 0
* Connection #0 to host dl.slic3r.org left intact
```

foreachthing, 2019-12-03:
Here's mine (from Switzerland with WIn10):
```
C:\Users\Me>curl -vL https://dl.slic3r.org/dev/
*   Trying 92.243.6.202...
* TCP_NODELAY set
* Connected to dl.slic3r.org (92.243.6.202) port 443 (#0)
* schannel: SSL/TLS connection with dl.slic3r.org port 443 (step 1/3)
* schannel: checking server certificate revocation
* schannel: sending initial handshake data: sending 178 bytes...
* schannel: sent initial handshake data: sent 178 bytes
* schannel: SSL/TLS connection with dl.slic3r.org port 443 (step 2/3)
* schannel: failed to receive handshake, need more data
* schannel: SSL/TLS connection with dl.slic3r.org port 443 (step 2/3)
* schannel: encrypted data got 2976
* schannel: encrypted data buffer: offset 2976 length 4096
* schannel: sending next handshake data: sending 126 bytes...
* schannel: SSL/TLS connection with dl.slic3r.org port 443 (step 2/3)
* schannel: encrypted data got 274
* schannel: encrypted data buffer: offset 274 length 4096
* schannel: SSL/TLS handshake complete
* schannel: SSL/TLS connection with dl.slic3r.org port 443 (step 3/3)
* schannel: stored credential handle in session cache
> GET /dev/ HTTP/1.1
> Host: dl.slic3r.org
> User-Agent: curl/7.55.1
> Accept: */*
>
* schannel: client wants to read 102400 bytes
* schannel: encdata_buffer resized 103424
* schannel: encrypted data buffer: offset 0 length 103424
* schannel: encrypted data got 2931
* schannel: encrypted data buffer: offset 2931 length 103424
* schannel: decrypted data length: 177
* schannel: decrypted data added: 177
* schannel: decrypted data cached: offset 177 length 102400
* schannel: encrypted data length: 2725
* schannel: encrypted data cached: offset 2725 length 103424
* schannel: decrypted data length: 613
* schannel: decrypted data added: 613
* schannel: decrypted data cached: offset 790 length 102400
* schannel: encrypted data length: 2083
* schannel: encrypted data cached: offset 2083 length 103424
* schannel: decrypted data length: 2
* schannel: decrypted data added: 2
* schannel: decrypted data cached: offset 792 length 102400
* schannel: encrypted data length: 2052
* schannel: encrypted data cached: offset 2052 length 103424
* schannel: decrypted data length: 1909
* schannel: decrypted data added: 1909
* schannel: decrypted data cached: offset 2701 length 102400
* schannel: encrypted data length: 114
* schannel: encrypted data cached: offset 114 length 103424
* schannel: decrypted data length: 2
* schannel: decrypted data added: 2
* schannel: decrypted data cached: offset 2703 length 102400
* schannel: encrypted data length: 83
* schannel: encrypted data cached: offset 83 length 103424
* schannel: decrypted data length: 20
* schannel: decrypted data added: 20
* schannel: decrypted data cached: offset 2723 length 102400
* schannel: encrypted data length: 34
* schannel: encrypted data cached: offset 34 length 103424
* schannel: decrypted data length: 5
* schannel: decrypted data added: 5
* schannel: decrypted data cached: offset 2728 length 102400
* schannel: encrypted data buffer: offset 0 length 103424
* schannel: decrypted data buffer: offset 2728 length 102400
* schannel: schannel_recv cleanup
* schannel: decrypted data returned 2728
* schannel: decrypted data buffer: offset 0 length 102400
[... 22 lines cut ...]
   <tr class="even"><td class="indexcolicon"><a href="/"><img src="/theme/icons/folder-home.png" alt="[PARENTDIR]" /></a></td><td class="indexcolname"><a href="/">Parent Directory</a></td><td class="indexcollastmod">&nbsp;</td><td class="indexcolsize">  - </td></tr>
   <tr class="odd"><td class="indexcolicon"><a href="win/"><img src="/theme/icons/folder.png" alt="[DIR]" /></a></td><td class="indexcolname"><a href="win/">win/</a></td><td class="indexcollastmod">2019-12-02 09:30  </td><td class="indexcolsize">  - </td></tr>
   <tr class="even"><td class="indexcolicon"><a href="linux/"><img src="/theme/icons/folder.png" alt="[DIR]" /></a></td><td class="indexcolname"><a href="linux/">linux/</a></td><td class="indexcollastmod">2019-12-02 09:30  </td><td class="indexcolsize">  - </td></tr>
   <tr class="odd"><td class="indexcolicon"><a href="mac/"><img src="/theme/icons/folder.png" alt="[DIR]" /></a></td><td class="indexcolname"><a href="mac/">mac/</a></td><td class="indexcollastmod">2019-09-24 00:30  </td><td class="indexcolsize">  - </td></tr>
</table>
        <!--
<div class="block">
                You can add your own HTML above or below the directory listing. Simply explore the <code>header.html</code> and <code>footer.html</code> files.
        </div>
-->
<!--/.postlisting-->

</div><!--/.wrapper-->

<div class="footer">
<!--    Apaxy by <a href="https://twitter.com/adamwhitcroft">@adamwhitcroft</a>
-->
</div><!--/.footer-->
<script type="text/javascript">
// grab the 2nd child and add the parent class. tr:nth-child(2)
document.getElementsByTagName('tr')[1].className = 'parent';
</script>
</body></html>
* Connection #0 to host dl.slic3r.org left intact
```

fitzterra, 2019-12-03:
Hmmm.... very interesting. Testing from DigitalOcean and AWS instances it works. Also from a box at another hosting provider in SA.

Trying it from behind my own ISP as well as from behind my employers ISP, I get the Forbidden error.

I'll take it up with my ISP to follow up. Thanks for the help looking into this @foreachthing  :+1:

closed 2019-12-03 by fitzterra

lordofhyphens, 2019-12-03:
@alranel can you check your blacklists? That is very likely what it is.