slic3r/Slic3r #4900: --export-obj available in command line

xyzdevops, 2019-12-03:
In the manual it says about this feature but in practice it doesn't work.

The help message returned by --help only lists --output-svg.

I am very confused why the manual thinks this is an already existing feature.

Version 1.3

lordofhyphens, 2019-12-03:
There are 2 CLI clients currently. One is c++ based, the other is perl

xyzdevops, 2019-12-04:
The readme should mention the perl one.

The manual should also be sectioned to clearly show the features of each CLI, currently it is highly confusing.

lordofhyphens, 2019-12-04:
slic3r/Slic3rManual awaits your pull requests.

xyzdevops, 2019-12-04:
I can't improve the document until I figure out how it works. Spent all day trying to dockerise the app so I can run the perl CLI to see if it will take this --export-obj flag.

My ultimate aim is to use the --cut-grid and the --export-obj so my derivative files are in a useful format.

lordofhyphens, 2019-12-04:
Why would you need to dockerize the application? The downloads from dl.slic3r.org/dev are all self contained and use the Perl interface (as the c++cli is experimental until support gets ported).

Even then, building the Perl side leverages local::lib as to not break your local system.

xyzdevops, 2019-12-04:
Because I want to run it in kubernetes.

Also I have Mac os Catalina so it wouldn't just run.

lordofhyphens, 2019-12-04:
For Catalina:

https://dl.slic3r.org/dev/mac/jenkins

No idea why the Travis builds don't work and the older Jenkins do on Mojave
and newer.

Have been trying on and off for months to figure that out.

xyzdevops, 2019-12-05:
Can someone please tell me if the --export-obj is a feature or not, as it doesn't seem to work on either CLI.

lordofhyphens, 2019-12-05:
I just checked with the C++ CLI.
````
./slic3r --export-obj <inputfilename>
````
should produce <inputfilename>.obj

You can check the CLI options in the code via `xs/src/libslic3r/PrintConfig.cpp` and via `slic3r --help`.

I just ran across a bug where apparently `stl_generate_shared_vertices()` in admesh gets caught in an infinite loop. I am currently looking into it.

lordofhyphens, 2019-12-05:
Unfortunately, apparently none of that has changed in years and so I am tracking down what could have broken.

lordofhyphens, 2019-12-05:
Ok, looks like the method expect the mesh to be repaired. I've committed an update which calls `repair()` on the mesh copy used for export.

lordofhyphens, 2019-12-05:
@xyzdevops update master branch and give it a try.

lordofhyphens, 2019-12-05:
@Oekn5w can you do a diff check around your code with the call to repair() removed in `IO.cpp:158`

to see if you can see if  the matrix reload changes broke anything?

xyzdevops, 2019-12-05:
I still get:
 Unknown option: export-obj

In the --help there is no such options for export as listed in the manual.

lordofhyphens, 2019-12-05:
On the c++ cli? On a checkout from master?

lordofhyphens, 2019-12-05:
Please do a fresh checkout of the master branch, build it with CMake.

xyzdevops, 2019-12-05:
`
FROM ubuntu:latest

RUN apt-get update
COPY sources.list /etc/apt/sources.list
RUN apt-get update
RUN apt-get install -y git

WORKDIR /app
RUN git clone https://github.com/Slic3r/Slic3r.git .

ENV LDLOADLIBS=-lstdc++
RUN apt-get install -y build-essential cpanminus liblocal-lib-perl cmake
RUN apt-get build-dep -y slic3r
RUN apt-get install -y libxmu-dev freeglut3-dev
RUN apt-get install -y libboost-thread-dev libboost-system-dev libboost-filesystem-dev
RUN apt-get install -y libextutils-parsexs-perl
RUN apt-get install -y libcurl4-openssl-dev

RUN mkdir build && cd build && cmake ../src && cmake --build . -- -j4

`

That Dockerfile should grab your latest changes.

I'm just testing the perl vs c++ without --export-object, then with it subsequently.

I will let you know what happens.

xyzdevops, 2019-12-05:
the c++ --cut-grid no longer works at all for my model, so it is a regression from my perspective.

the c++ '--export-obj --cut-grid...' does nothing, which is better than rejecting the command like the perl CLI does.

Oekn5w, 2019-12-05:
I'll look into the mesh flattening in the model's function again, didn't find anything briefly.

But I'm not quite following the conversation here, is this about differences between perl and c++ CLI, or the export not working?

If the latter and the repair fixes that, then I'd like to know how to reproduce (although, to be fair, I didn't actually try anything, today at least)

xyzdevops, 2019-12-05:
The comparison between languages is to sanity check what the state of some features are; to isolate bugs

lordofhyphens, 2019-12-05:
Export obj is NOT in the Perl side. I don't intend to port it.

lordofhyphens, 2019-12-05:
@Oekn5w I just tried to export obj with a 20mm stl box from the c++ cli.

Infinite loop inside shared vertex generation.