slic3r/Slic3r #4901: Support Material not printed on the first layer

danielalexe, 2019-12-06:
**Describe the bug**
The support material is not printed for the first layer and as a result the support is printed in the air.

**To Reproduce**
Steps to reproduce the behavior:
1. Upload the model spool_holder_1.stl
2. Go to print settings
3. Go to support material
4. Check Generate support material
5. Set Raft Layers at 0
6. See the preview of the print on the plater tab (0.3 -> 0.6)

**3D Model and Slic3r Configuration Export**
[spool_holder_1.zip](https://github.com/slic3r/Slic3r/files/3932451/spool_holder_1.zip)

**Expected behavior**
Support material is printed from the first layer.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 1.3.0

**Additional context**
I have tried every combination of settings in the support material section and the only way to print the model with the support material is to add 1 layer of raft to the model and I do not want that.

lordofhyphens, 2019-12-06:
Just to be sure, have you checked with a build from dl.slic3r.org/dev/win ?

danielalexe, 2019-12-07:
Used Slic3r-master.2019.12.05.2373.c2aa7e7b.32bit-PR
Same behavior at 0.3 (first layer) no support material is generated.

lordofhyphens, 2019-12-07:
Thanks for checking.

hubertus-3d, 2019-12-28:
Hallo You two,
as the third person dealing with thies problem (also on teh user side) I can confirm having the probably same issue on my equipment:
since 2 months I am digging into 3D printing. I bought a Renkforce RF2000 printer, which included a software package (RepetierHost V1.6.2 with Slic3r V1.2.9 as one of the Slicer-engines).
Up to now I only used one filament for printing, which means extruder #2 has been inactive so far. In order to have support material easily removed at complex part geometries, I moved on to use the second extruder for water soluable filament. I proved the operation of this new filament at extruder #1.
Unfortunately extruder #2 does not output reasonable filament volume, while printing - just very thin worm. I have controled, that Slic3r generates the relevant G-Code, which it does.
BUT: The values for the filament advance G-Code (Ex.xxx) seem to be to small by the factor of 10, if compared with the ones for extruder #1. As a result just very few filament exits the nozzle, which does neither glue to the printbed, nor generate a meaningfull structure.
Remarkable: the starting line at the front of the print bed seems to be printed correctly.
On my PC (Trekstor Primebook C13) I have installed Windows 10 Home64 with all available Updates.
When I looked for support I found a two year old forum entry. That user estimated a malcalculation of the advance-factor of ca.5.3 (may be a missing 2Pi operation when calculating the filament length to be advanced?).
I already tried to solve the prblem by installing the latest version 2.1.6 of RepetierHost including SLic3r version 1.3..1b. Unfortunately the problem still exists- so it seems to be a long existing problem.
Also the Repetier firmware of the printer has been updated from 1.10 to 1.39. This did not change anything neither.
comment 1: the problem keeps existing, when I call Slic3r directly (not using RepetierHost as GUI)
comment 2: when I use CURA as slicer engine inside RepetierHost this problem does not exist (but other problems hinder me using this as workaround).
I hope this clears the reason for the problem, and makes possible a short term fixing - which I would appreciate.
Please feel fre to contact me for further information or tests.
Have  a good transition to the new year, and hopefully a few days of vacation.
Erst mal allen noch ein paar ruhige Tage zum Jahreswechsel

himukai-takumi, 2020-04-02:
I had the same problem. (Slic3r: ver 1.3.1-dev)
Stopping using Brim no longer causes support to float in the air.

In my opinion, there is a problem in judging a cylindrical 3D model.
I look forward to making a good contribution to this case.

[image]
[image]

danielalexe, 2020-04-02:
Arigato Himukai Takumi
I have tested the model without the brim and it is as you said, the support is being generated correctly.
Now I tend to believe that both the brim and the support material are linked together and they interfere with one another.
I have made a couple of tests and I saw that the lower the brim the more support material is generated inside the cylinder model.
It is possible that the support material thinks that there is brim inside the cylinder and that is why it does not generate it properly. Thus even though there is no visible brim in the preview, probably cleaned up by some code, the clean up is done after the support material is generated and that is what causes the issue.

GlennFR, 2021-09-25:
Hello, I have been fighting this issue for several days, I'm using some PET-G filament that really has to be squished at the first layer, otherwise it just won't stick.  I also use brims, as my parts are hollow and complex.  I hadn't noticed the link between brims and support material - I'll try a test. **BUT** in any case I'm surprised that this case isn't assigned and that not many people suffer...   I've trashed almost a half spool of filament already....
Please, somebody ?

himukai-takumi, 2021-09-25:
I haven't tried it, but I have two ideas.
-Include a brim in the model.
-The other is to avoid holes by tilting the model.  

I hope this helps.