slic3r/Slic3r #4903: Where is the configuration "Show incompatible print and filament presets"?

Salty-Doug, 2019-12-12:
### Version
1.3.0 

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_ Mac OS 10.14.6 _

### Behavior
This bug was reported in Issue 4563. 
https://github.com/slic3r/Slic3r/issues/4563

It was closed after the OC found the configuration item "Show incompatible print and filament presets" was checked. I am unable to find this configuration item.  Here is a screen shot of the Prefernces panel. 
[image]

I have also looked in /users/{my name}/library/application support/slic3r
Any help will be appreciated.

Salty Doug

foreachthing, 2019-12-12:
> _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. _Only_ use normal Slic3r version IDs._

[image]

There's nothing in the [manual](https://manual.slic3r.org/configuration-organization/configuration-organization) about "Show incompatible print and filament presets" ... ?! I can't remember that setting in Slic3r - only Prusa.

Salty-Doug, 2019-12-13:
Thanks for the reply.
I am using Slic3r 1.3.0 and the Print Settings presets are not showing on my Mac 10.14.6.
Today I see that the issue was for Prusa software in my Google search. Sorry.

[image]
[image]
Can you help me?

Best
Salty Doug

foreachthing, 2019-12-13:
um ... ok! :-o
Maybe export config bundle (Slic3r - File - Export Config Bundle...) and load (Slic3r - File - Load Config Bundle..) 'em again.
Make backups first!!!

Salty-Doug, 2019-12-14:
Ok. Thanks I'll try that.