slic3r/Slic3r #4907: LoadLibrary failed with error 87: The parameter is incorrect.

billsoranno, 2019-12-26:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
I downloaded Slic3r-1.3.0.64bit.zip. I unzipped the file. When I double click on Slic3r.exe, I receive the error in the title.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to folder the zip file was unzipped to.
2. Double Click on Slic3r.exe
3. See error dialog box.

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.
I cannot do this since I cannot even start the application.

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect the application to open so I can load an stl file and slice it.
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version 1809
     - _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
1.3.0
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

**Additional context**
Add any other context about the problem here.

billsoranno, 2020-01-24:
I am still having this problem, is anyone monitoring this product?

lordofhyphens, 2020-01-24:
Can't reproduce. Did you check with 1.3.1-dev build from dl.slic3r.org/dev/win

lordofhyphens, 2020-01-24:
Also this seems to be video driver related.

billsoranno, 2020-01-24:
Joseph,
Thank you for the replies. I will download version 1.3.1 this evening.

Bill Soranno