slic3r/Slic3r #4908: Installing Wx failed

nonplanarwt, 2019-12-28:
during my installation i`ll get the following message:

--> Working on Wx::build::MakeMaker
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9932.tar.gz ... OK
Configuring Wx-0.9932 ... OK
Building and testing Wx-0.9932 ... FAIL
! Installing Wx::build::MakeMaker failed. See /root/.cpanm/work/1577534811.19367/build.log for details. Retry with --force to force install it.
! Installing the dependencies failed: Module 'Wx::build::MakeMaker' is not installed
! Bailing out the installation for Wx-GLCanvas-0.09.
Don't worry, this module is optional.
The following prerequisites failed to install: Wx

i tried to install on my win10 and ubuntu-orangepie and i get the same failure on both..

the logfile says that 'WX' is not a known MakeMaker parameter name.

can anyone help

lordofhyphens, 2019-12-28:
If you aren't modifying/developing Slic3r, use a build from dl.slic3r.org/dev 

Wxperl is something I am trying to get away from and it isn't something we maintain here.

nonplanarwt, 2019-12-28:
i tried to brew a slicer version for nonplanar slicing as the instruction manual says

lordofhyphens, 2019-12-28:
If brewing you can probably steal the local-lib directory from an archive

lordofhyphens, 2019-12-28:
The wxperl dependency is probably the worst part about doing a from source build of Slic3r

nonplanarwt, 2019-12-28:
maybe i forgot one step??
this was my way:
1. install Slic3r perl 5.24.1
2. install the boost libaries 1_63_0
3. getting the prebuild perl- dependencies
4. git clone the slic3r from alexrj
5. extract the dependencies (local-lib) in root folder of slic3r
6. perl build.pl and when is start the perl build.pl --gui the message appears....

lordofhyphens, 2019-12-28:
Do you have the wx development libraries installed?

The --gui step is completely separate from everything else for the build process (it only installs prerequisites, not anything actually part of Slic3r)

wjk199511140034, 2019-12-29:
I recommend Citrus Perl instead of Strawberry Perl
install Citrus Perl 5.24.1(just skip set GCC)
install the boost libaries 1_63_0
git clone the slic3r from alexrj
perl build.pl (maybe you need `cpan JSON App::cpanminus` and `cpanm local : : lib` during this step)
and then perl build.pl --gui 
Citrus Perl is very fast than Strawberry Perl in every single process!

wjk199511140034, 2019-12-29:
I recommend Citrus Perl instead of Strawberry Perl
install Citrus Perl 5.24.1(just skip set GCC)
install the boost libaries 1_63_0
git clone the slic3r from alexrj
perl build.pl (you may need `cpan JSON App::cpanminus` and `cpanm local::lib` during this step)
and then perl build.pl --gui
Citrus Perl is very fast than Strawberry Perl in every single process!
------------------------------------------------------------------------
try `cpanm --force Module::Name` if you insist Strawberry

lordofhyphens, 2019-12-29:
If you are using Strawberry, then just steal the local-lib directory from a
release.

gdville, 2020-06-13:
@lordofhyphens could you please be more explicit on an example of where to find local-lib directory release?

lordofhyphens, 2020-06-13:
@gdville every release build of Slic3r has a local-lib directory in it. So long as you use the same Perl as was used to build Slic3r, you can copy this folder.

Kramer84, 2021-01-14:
I had a similar problem trying to compile the GUI part on Ubuntu 20.04, as the pre-compiled GUI was unreadable, I found a work around by manually installing wx and wx-perl : 
```
sudo apt-get install libwxgtk3.0-gtk3-0v5
sudo apt-get install libwxbase3.0-0v5
sudo apt-get install libwx-perl
```
I still get some errors : 
```
App::cpanminus is up to date. (1.7044)
Class::Accessor is up to date. (0.51)
Growl::GNTP is up to date. (0.21)
--> Working on LWP::UserAgent
Fetching http://www.cpan.org/authors/id/O/OA/OALDERS/libwww-perl-6.52.tar.gz ... OK
Configuring libwww-perl-6.52 ... OK
Building and testing libwww-perl-6.52 ... FAIL
! Installing LWP::UserAgent failed. See /home/motherloadubu/.cpanm/work/1610586231.283842/build.log for details. Retry with --force to force install it.
Don't worry, this module is optional.
Net::Bonjour is up to date. (0.96)
OpenGL is up to date. (0.70)
Socket is up to date. (2.031)
Wx is up to date. (0.9932)
--> Working on Wx::GLCanvas
Fetching http://www.cpan.org/authors/id/M/MB/MBARBON/Wx-GLCanvas-0.09.tar.gz ... OK
Configuring Wx-GLCanvas-0.09 ... OK
Building and testing Wx-GLCanvas-0.09 ... OK
Successfully installed Wx-GLCanvas-0.09
1 distribution installed
```
But the GUI seems to be working.

For the libwww-perl-6.52 FAIL, installing manually the package did not solve this issue, as for my machine the latest version for this package seems to be 6.43-1

busyBeaver1, 2021-03-29:
@Kramer84, thank you very much! I had the same problem and it helped me.