slic3r/Slic3r #4909: [Optimization] Bad Dijkstra implementation

ilovelinux, 2019-12-29:
**Describe the bug**
A clear and concise description of what the bug is. The complexity of your implementation of the Dijkstra algorithm is O(n^2). It can be implemented with a complexity of O(n log n) using a priority queue/min heap instead of an array.

**To Reproduce**
Have a look here: https://github.com/slic3r/Slic3r/blob/354b6230bdea90b50b1c422a8da94c1f59313954/lib/Slic3r/GCode/MotionPlanner.pm#L253

**Expected behavior**
Use a priority queue/min heap.

Btw why do you use Perl? C++ has a lot of well-implemented data structures and is way more efficient

lordofhyphens, 2019-12-29:
Pretty sure that was already implemented in C++, but if you wish to submit a pull request with the changes I will take a look and probably approve it.

ledvinap, 2019-12-30:
https://github.com/slic3r/Slic3r/pull/2086

Sorry, the code needs some polishing. I hope I'll get on it, but I'm a bit too busy lately ...

ilovelinux, 2019-12-30:
> Pretty sure that was already implemented in C++, but if you wish to submit a pull request with the changes I will take a look and probably approve it.

I found this:
https://github.com/slic3r/Slic3r/blob/354b6230bdea90b50b1c422a8da94c1f59313954/xs/src/libslic3r/MotionPlanner.cpp#L84
so perl implementation is useless?