slic3r/Slic3r #4914: Request: Purge Line Setting

Kraligor, 2020-01-06:
**Is your feature request related to a problem? Please describe.**
Leaky extruders combined with a stringy filament make using a skirt for priming the extruder difficult. When it first starts extruding, there is a string of filament that doesn't adhere to the bed, possibly laying in the print zone, disturbing the actual print.

**Describe the solution you'd like**
A simple purge line is the best way to get around this issue. It is easily implemented too--all that is needed would be a third option next to brim and skirt with a few parameters such as position and speed settings.

**Describe alternatives you've considered**
My printer is calibrated, so the problem can't be fixed by tuning. I used to snip off the string while printing, which is not an optimal solution.

Right now, I'm using a purge line per custom g-code:
```M82 ; Extruder in absolute mode
G28; Home extruder
G1 X0 Y0 Z5 F{Z_TRAVEL_SPEED}
M107 ; Turn off fan
G90 ; Absolute positioning
M104 S150  ; start extruder heating
M190 S[first_layer_bed_temperature] ; set bed temp
M140 S[first_layer_bed_temperature]  ;wait for bed temp
M109 S[first_layer_temperature_0]  ; Wait for all used extruders to reach temperature
G92 E0 ; Reset extruder position
G1 X5 Y5 Z0.25 F1000.0 ; go intro start pos
G1 X60.0 E9.0 F600.0 ; intro line to prime extruder
G1 X100.0 E16 F800.0 ; intro line
G92 E0.0 ; reset extruder distance position
```

Kraligor, 2020-01-06:
If you want me to do it, I'm sure I can code it together, btw.

ledvinap, 2020-01-06:
@Kraligor : What's wrong with custom Gcode ?

Kraligor, 2020-01-06:
Oh, nothing per se. But I'd imagine it's a not too uncommon issue, and people who don't want to fiddle around with Gcode maybe don't even know that there's a solution to their problem.

VanessaE, 2020-01-07:
I do a prime line plus a 1-loop skirt (to bleed off any excess pressure).  For the prime line:

```gcode
M104 S[first_layer_temperature] ; start heating the hot end so that it heats in parallel
M140 S[first_layer_bed_temperature] ; start heating the bed
M109 S[first_layer_temperature] ; Wait for the extruder to be at temp
M190 S[first_layer_bed_temperature] ; Wait for the bed to be at temp
G28
M420 S1 ; restore bed level data
; Draw prime line
G92 E0
G1 Z10
G1 X190 Y2 F12000
G1 Z0.4
G1 X10 Y2 E30 F1200
G92 E0
```

The temperature commands are needed, to stall the print (and thus the prime line) until the hotend and bed are hot.

lordofhyphens, 2020-01-08:
@Kraligor if you want to take a swing at implementing this I'll look at a PR.

Kraligor, 2020-01-10:
Cool, will see if I can get something done this weekend.