slic3r/Slic3r #4915: Gyroid infill issue with an STL model

faeluke, 2020-01-10:
**Describe the bug**
Slic3r 1.3 crashes with the Gyroid pattern selected with the attached STL file.

**To Reproduce**
1. Load the model
2. Load the configuration file config.ini
3. If background processing is set in preferences Slic3r will crash
If background processing is not set, and save to gcode is selected, Slic3r will crash
I had selected background 
A clear and concise description of what the bug is.

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
No crash

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version [1.3.0
[bug.txt](https://github.com/slic3r/Slic3r/files/4043726/bug.txt)

**Additional context**
I was configuring for my device for the first time. I use Gyroid in Cura all the time. I wanted to try Slic3r due to extended print times and excessive non print moves in the generated g code.

[Archive.zip](https://github.com/slic3r/Slic3r/files/4043685/Archive.zip)

lordofhyphens, 2020-01-10:
@faeluke please try a build from https://dl.slic3r.org/dev/mac/jenkins/. 

I was unable to reproduce on a 1.3.1-dev build

faeluke, 2020-01-11:
V 1.3.1 did not crash. Thanks for the quick reply!

closed 2020-01-11 by faeluke