slic3r/Slic3r #4921: Slic3r crashes on gyroid infill

smartynov, 2020-01-25:
**Describe the bug**
Slic3r crashes when computing preview with Gyroid infill pattern, even for a primitive sphere.

**To Reproduce**
Steps to reproduce the behavior:
1. Open attached `sphere50.stl` model 
2. Set infill pattern to `Gyroid`
3. Click on Preview 
4. In a few moments Slic3r crashes

**3D Model and Slic3r Configuration Export**
[infill-gyroid-issue.zip](https://github.com/slic3r/Slic3r/files/4111386/infill-gyroid-issue.zip)

**Expected behavior**
I expect Slic3r not to crash and generate a preview with gyroid infill.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: 1.3.0

**Additional context**
I noticed this issue on a complex model and reduced to a single primitive. The `sphere.stl` attached is created in OpenSCAD as `sphere(50)`. At the same time (and same config) it manages to compute gyroid infill for cube primitive.

lordofhyphens, 2020-01-25:
https://dl.slic3r.org/dev/mac

smartynov, 2020-01-26:
Is there any particular version when this bug was fixed or should I try all of them? 

I tried `Slic3r-master-latest.dmg` so far and it does not work for me at all. I load STL and when I click on Preview – it just shows empty grid and "Cancel" button in the bottom bar – and nothing else happens. It does not crash, it does not use CPU – just waits. I was unable to get it working even with a cube STL.

lordofhyphens, 2020-01-26:
`latest` is generally fine.
Here it is on Linux with gyroid infill.
[image]

lordofhyphens, 2020-01-26:
and here it is on my Mojave Mac mini using your file and what should be your config:

[image]

lordofhyphens, 2020-01-26:
As a further test, I edited `fill_pattern` in the exported ini file to `gyroid` and then loaded that. Still sliced just fine with your exported sphere on https://dl.slic3r.org/dev/mac/jenkins/slic3r-1.3.1-dev-c2aa7e7b-osx.dmg

If you are using Mojave or newer MacOS, you should be using the newest build in the `jenkins` folder.

smartynov, 2020-01-26:
I am using a pretty old MacOS Sierra 10.12.6 (16G2136). I tried the latest file from Jenkins folder (slic3r-1.3.1-dev-c2aa7e7b-osx.dmg) and it works for me fine. I can generate gyroid on sphere and more complex objects that failed before.

The only thing that bothers me is that jenkins builds are not signed (and probably not very stable as dev versions). Are there any plans on releasing version 1.3.1?

closed 2020-01-26 by smartynov

lordofhyphens, 2020-01-26:
Sigh, they should be signed. I need to update the keys on that build server then. @alranel ?

I don't usually do much to break existing stuff, so the answer is "when it is worth doing compared to 1.3.0"