slic3r/Slic3r #4922: Post Processing Script is not executed

joe-hidden, 2020-01-26:
Version of Slicer: Daily Build Slic3r-master.2019.12.05.2373.c2aa7e7b.32bit-PR
OS: WIN7 Patched to date

Script defined in Post-Processing Scripts is not executed
tested with Full Qualifed Pathnamed .exe, .cmd and .bat files.

Executing the .bat, .cmd or .exe with cut&past in a CMD Window at c:\ is working when filename is attached

just calling a .bat or .cmd consiting of only "echo test > c:\temp\test.txt" will not create the test file. (c:\temp exits)

To Reproduce
Steps to reproduce the behavior:

Run Slic3r 
Go to 'print settings and add a custom post processing script'
Load any model.
Click on 'Generate G-Code'

Expected behavior
I would expect the post-processing code to be executed.

lordofhyphens, 2020-01-26:
I don't think I have ever actually tested using batch files or executable files on Windows.

lordofhyphens, 2020-01-26:
Are you sure you should be using a 32bit version of the application and not 64?

foreachthing, 2020-01-27:
Latest 64bit build from https://dl.slic3r.org/dev/win/ works on Win 10 x64, though.
I am using an .exe, no additional parameters.

joe-hidden, 2020-01-29:
Well, it was perfectly working with 1.2.9

joe-hidden, 2020-01-29:
i tried both with same result in the mean time

joe-hidden, 2020-02-01:
even using the newest dev version and trying to start C:\Windows\System32\cmd.exe 
so something is strange here

lordofhyphens, 2020-02-03:
No, should not matter at all. I am still suspecting something on your local environment (the code around that hasn't changed since 1.2.9)

joe-hidden, 2020-02-03:
i am willing to look into, but how?