slic3r/Slic3r #4923: Grid infill shouldn't break the infill lines on intersections

l29ah, 2020-01-28:
**Is your feature request related to a problem? Please describe.**
Nowadays Slic3r stops extruding when a Grid infill line crosses another line, resulting in voids that compromise the part rigidity for no good reason.

**Describe the solution you'd like**
Grid infill shouldn't stop extruding when crossing an existing Grid infill line.

**Describe alternatives you've considered**
Making it configurable doesn't seem to make sense since Grid is not used at 100% infill, otherwise there's always space for extra material.

lordofhyphens, 2020-01-28:
Is this the rectilinear infill mode?

l29ah, 2020-01-28:
No, as i mentioned five times already, it's the Grid infill mode.

l29ah, 2020-01-29:
[image]
Arrows where underfills are directly observable, dots where they are present but hidden by other walls.

lordofhyphens, 2020-01-29:
So here's gcode.ws rendering of a layer of a 20mm cube @ around 3% infill (mostly to make it look easier). 

[image]
[image]
[image]
[image]
[image]
[image]

I think what you are seeing is Slic3r dragging through the existing plastic.

lordofhyphens, 2020-01-29:
Same cube, 15% infill. Note that Slic3r happily crosses lines. 

[image]
[image]
[image]
[image]
[image]

ledvinap, 2020-01-29:
This is IMO problem with PET. When crossing perpendicular infil, extrudate gets stripped from nozzle and there is small void just behind.
Alternating infill direction between layers will probably help a lot.

bubnikv, 2020-01-29:
It happens with any material if you print quickly enough. I suppose it may
have something to do with the non-Newtonian properties of the molten
polymer: The flow is stopped when crossing an existing line, and then it
takes time before the plastic flows with the flow velocity before crossing.
This is a general issue visible not only on the infill, but for example
when filling in the top surfaces with a slight intentional overfill to fill
in the corners between the perimeters and the infill lines, where the flow
is constrained by the already extruded material. One than may see
underextrusion on the following extrusion, if the following extrusion
happens into a free enough space.

Trellian, 2020-01-29:
> This is IMO problem with PET. When crossing perpendicular infil, extrudate gets stripped from nozzle and there is small void just behind.
> Alternating infill direction between layers will probably help a lot.

I have also experienced this a lot with PET, very little with ABS.