slic3r/Slic3r #4924: Brief nozzle change in between layers when printing with multi nozzles/materials

Lucent-RFTG, 2020-01-30:
**Version**
slic3r version: 1.3.0

**Printer**
Firmware type: marlin
Printer: Lulzbot Taz 6

**Describe the bug**
I am trying to print a component with two different materials. They are extruded through two individual nozzles. There are some layers which are made from one material only. This is done with both extruder **0** and **1** respectively. When printing with extruder 1 I find that after the layer change slic3r changes briefly to nozzle **0**, with the following G-code,
...
; AFTER LAYER CHANGE
T0
G92 E0
T1
G92 E0
...
and then back to **1**.

Note: This is only an issue when printing with extruder **1**.

The issue is that the materials are cooled down to prevent oozing which results in the printer having to heat/cool the nozzles at each layer. it furthermore affects the print quality as such as there is quite a bit of oozing.

**3D Model and Slic3r Configuration Export**
Attached to this report is a the model and the config file

**Expected behavior**
I would not expect the nozzle to be switched when printing with one material only, in a multi material environment, like it does when printing with extruder **1**.

**Desktop (please complete the following information):**
- I tried this in various versions of slic3r. This issue seems to be present in all recent versions
- I tired it on Mac OS and Windows

**Attached are the following files**
[slic3r issue.zip](https://github.com/slic3r/Slic3r/files/4135135/slic3r.issue.zip)
[image]

lordofhyphens, 2020-01-30:
I believe that Slic3r assumes that it always starts with extruder 0 at each layer. I do not recall why that is an assumption it has to make (e.g. why it cannot track the active extruder across layers).

Lucent-RFTG, 2020-01-31:
> I believe that Slic3r assumes that it always starts with extruder 0 at each layer. I do not recall why that is an assumption it has to make (e.g. why it cannot track the active extruder across layers).

Thanks very much for your response. I can assume why one would code it this way -- most prints would be carried out using either a prime tower for bot extruders and/or using the multi material option on every single layer.
However, it would be good to have some kind of function within the source code which detects whether there are two consecutive layers with one material only.
I had a look at one of your post-processor scripts. Unfortunately it removes all tool changes.