slic3r/Slic3r #4926: Perimeters not touching each other

jes, 2020-02-06:
**Describe the bug**
On all layers except the bottom layer, some perimeters are not touching each other, leading to weak parts.

**To Reproduce**
I have noticed this in various places on various parts for quite a while. I modelled a small button to have lots of perimeters visible on the top layer and printed it out. The bottom layer came out perfectly, as intended, but all other layers seem to have perimeters that are too narrow. It's almost as if Slic3r is using a fat extrusion width to decide the spacing between extrusions, but a narrower extrusion width when actually extruding.

**3D Model and Slic3r Configuration Export**
[perimeters-not-touching.zip](https://github.com/slic3r/Slic3r/files/4164614/perimeters-not-touching.zip)

**Expected behavior**
I expected the perimeters to be touching each other.

**Screenshots**
https://img.jes.xxx/2542 - image of top layer showing outer perimeters not touching
https://img.jes.xxx/2543 - image of part held up to window showing light passing through (this is printed in Nylon so although the bottom layer is good, it still passes light as it is only 1 layer thick)
https://img.jes.xxx/2544 - image of bottom layer

**Desktop (please complete the following information):**
 - OS: Ubuntu 19.10
 - Version 1.3.1-dev (54a31ee)

lordofhyphens, 2020-02-06:
Screenshot of Slic3r's own 3d preview?

jes, 2020-02-06:
https://img.jes.xxx/2545

jes, 2020-02-06:
Might we be able to eliminate (or prove) machine problems if somebody could post a piece of GCode that prints properly on their machine, and I'll just adjust the temperatures and try to print it on mine?

I suspect it is caused by some combination of configuration settings I have, but I have no idea what it could be.