slic3r/Slic3r #4928: using command line option --export-sla-svg results in Unknown option: export-sla-svg

mmiscool, 2020-02-10:
in the docs it describes command line usage for exporting svg for SLA.
https://manual.slic3r.org/advanced/command-line

> --export-sla-svg (shortcut: --sla): slices the given model(s) and exports slices for SLA printers as SVG files (with infill patterns, raft support...)

Tried on 1.3.1-dev also.

**Describe the bug**
Running the following on debian 9 with latest 1.3.0 build
./Slic3r --export-sla-svg --layer-height 0.2 test.stl

also tried 
./Slic3r --sla --layer-height 0.2 test.stl
./Slic3r --export-sla-svg --layer-height 0.2 test.stl

Result:
Unknown option: export-sla-svg

**To Reproduce**
./Slic3r --export-sla-svg --layer-height 0.2 test.stl

**Expected behavior**
Getting a SVG file of the sliced object including support material.

lordofhyphens, 2020-02-11:
That's the c++ cli

mmiscool, 2020-02-11:
> That's the c++ cli

What dose that mean?

lordofhyphens, 2020-02-12:
There's a separate application that is being worked on as part of the effort to fully remove Perl. It has a slightly better CLI.

mmiscool, 2020-02-12:
is there a binary for that and where might  it be available for linux.

elsatch, 2020-10-19:
I have just tried with Slic3r versions:

- 0.6.0 (Doesn't have command line version)
- 0.7.0
- 0.7.1
- 0.7.2b
- 0.8.3
- 0.8.4
- 1.0.1
- 1.1.2 (experimental)
- 1.1.7
- 1.2.9a
- 1.3.0

The result is always the same: Unknown option: export-sla-svg

I assume this function was never implemented as per the manual. In this [thread](https://github.com/slic3r/Slic3r/issues/4578) @alranel suggests to use Slice to SVG to get the paths, but it seems it's not possible to export the FDM G-Code toolpaths (yet!)