slic3r/Slic3r #4932: Wide skirt?

curnowdj266, 2020-02-17:
### Version
Slic3r-1.3.0-x86_64.AppImage

### Operating system type + version
Ubuntu 18.04 Linux
### Behavior
In settings I set the skirt to 3mm but when the model is printed the skirt comes out a 12mm
Is there something I am missing?
[image]

TD5023, 2020-02-18:
The 3mm you have set is the distance from the inside of the skirt to the outer shell of the object, not the width of the skirt itself. The driving force behind your current skirt generation is the 100mm minimum extrusion length, which will make the skirt take as many loops as necessary to extrude 100mm of filament. Also, just in case it wasn't clear, brim width has nothing to do with the skirt. 

I personally like using minimum extrusion length as the driver since it will guarantee that you're extruding cleanly by the time the code is done with the skirt, but 100mm is a bit excessive. I use 50, but I also start with my filament mostly retracted. If you're getting a 12mm skirt, try a smaller number like 20 or 30 (maybe even less still).

lordofhyphens, 2020-02-18:
Or use skirt loops if you have to have it wider

curnowdj266, 2020-02-18:
Ok. Thanks that explains all.

closed 2020-02-18 by curnowdj266