slic3r/Slic3r #4935: Use mingw build slic3r command line in woindows

wjk199511140034, 2020-02-20:
Try to compile slic3r CLI (cpp) with cmake + mingw
An error occurred at last, and the compiled exe seem nonfunctional.
This is the error message
[image]

full log is here
[cmd.txt](https://github.com/slic3r/Slic3r/files/4230802/cmd.txt)

compile environment:
win10 x64 
cmake 3.16.2
mingw w64  4.3.5
boost_1_63_0
wxWidgets 3.0.0

lordofhyphens, 2020-02-20:
There is an additional compiled piece needed on windows, and I believe that
CMake files don't account for mingw when deciding to build it.

wjk199511140034, 2020-02-21:
> There is an additional compiled piece needed on windows, and I believe that CMake files don't account for mingw when deciding to build it.
> […](#)
I don't know what additional compiled piece is, I built it based on "Running-Slic3r-from-git-on-Windows" and file in "Slic3r/src/windows-build.txt",  there was many errors in configure and generate steps, thanks Google for helping me solve them one by one, but in the end, I couldn't get a valid information and log T_T. 
I think bypassing perl can help me understand slic3r, after all I'm an amateur coder.

lordofhyphens, 2020-02-21:
https://github.com/slic3r/Slic3r/blob/master/src/CMakeLists.txt#L501

https://github.com/slic3r/Slic3r/blob/master/src/CMakeLists.txt#L505

Change that if statement to if MSVC OR MINGW and see where you get.

wjk199511140034, 2020-02-22:
I have a little confused on the L501 and L504, if you mean the 499 and 503 it made sense, But its make  no different I got the same error at last step.
....

wjk199511140034, 2020-02-25:
> https://github.com/slic3r/Slic3r/blob/master/src/CMakeLists.txt#L501
> 
> https://github.com/slic3r/Slic3r/blob/master/src/CMakeLists.txt#L505
> 
> Change that if statement to if MSVC OR MINGW and see where you get.

[image]

wjk199511140034, 2020-02-28:
To avoid this error, don't build slic3r_test or, just add //before boost :: nowide :: cerr in AMF.cpp line461and TMF.cpp line292.
But that was just self-deceive....

Other issue:
Run with `slic3r -g test.stl` or `slic3r -g box.3mf`
```
C:\dev\Slic3r\Build>slic3r.exe -gcode test.stl
CLI   ERR: Arrange: 1, center: 1
Infilling layers
Processing triangulated mesh
Preparing infill
Generating skirt
Generating brim
Exporting G-Code...
CLI   ERR: Failed to remove the output G-code file from test.gcode.tmp to test.gcode. Is test.gcode.tmp locked?
```
```
C:\dev\Slic3r\Build>slic3r.exe -gcode box.3mf
CLI   ERR: Arrange: 1, center: 1
Infilling layers
Processing triangulated mesh
Preparing infill
Generating skirt
Generating brim
Exporting G-Code...
CLI   ERR: Failed to remove the output G-code file from box.gcode.tmp to box.gcode. Is box.gcode.tmp locked?
```
Don't know what's CLI   ERR: Arrange: 1, center: 1
But seem like this *.gcode.tmp are working fine if I change it to *.gcode，
I suspected that is permission issue, tried moving to another disk, and run with admin, all dead..

TLAS11, 2020-03-15:
> Try to compile slic3r CLI (cpp) with cmake + mingw
> An error occurred at last, and the compiled exe seem nonfunctional.
> This is the error message
> [image]
>
> full log is here [...]

I think I overcame this error in Windows with Visual Studio.  Make sure you include the following library in the slic3r_test:
Release\boost-nowide.lib

wjk199511140034, 2020-03-16:
> I think I overcame this error in Windows with Visual Studio. Make sure you include the following library in the slic3r_test:
> Release\boost-nowide.lib

 I overcame this error by delete slic3r_test lol