slic3r/Slic3r #4939: Rotate "Around z axis" appears absolute (dev release)

DocTrucker, 2020-02-26:
Slic3r: 1.3.1-dev-54a31ee
OS: Ubuntu 18.04.4

Load STL
Select model.
Right click - Rotate - Around z axis...
Input 90 - press ok
Model moves 90 degrees counter clock wise as expected.
Input 90 - press ok
BUG Model doesn't move.
Input 0 - press ok
Model doesn't move. (As expected for relative rotations, but not absolute.)
Input -90 - press ok
BUG Model rotates 180 degrees.

Around x and y axis work as expected other than the previously reported bug about rotations of -1.

dwillmore, 2020-02-27:
I can reproduce this.  Put a value in when rotating around X with the dialog and you get relative rotations.  But the Z dialog values appear to be absolute, not relative.

msquare-2, 2021-02-15:
I would like to participate in this issue, I am new to the opensource contribution and have a good command of C++. Can you guys help me with how to get started?