slic3r/Slic3r #4946: Slic3r - AppImage not working at all !

alfsch1, 2020-03-19:
Describe the bug
start appimage...nothing. then test in shell, errors shown:
$ ./Slic3r-1.3.1-dev-54a31ee-x86_64.AppImage libGL error: unable to load driver: radeonsi_dri.so libGL error: driver pointer missing libGL error: failed to load driver: radeonsi libGL error: unable to load driver: radeonsi_dri.so libGL error: driver pointer missing libGL error: failed to load driver: radeonsi libGL error: unable to load driver: swrast_dri.so libGL error: failed to load driver: swrast The program 'slic3r.pl' received an X Window System error. This probably reflects a bug in the program. The error was 'BadValue (integer parameter out of range for operation)'. (Details: serial 575 error_code 2 request_code 155 minor_code 24) (Note to programmers: normally, X errors are reported asynchronously; that is, you will receive the error a while after causing it. To debug your program, run it with the --sync command line option to change this behavior. You can then get a meaningful backtrace from your debugger if you break on the gdk_x_error() function.)

To Reproduce
Steps to reproduce the behavior:
system: -Version-
Kernel : Linux 4.19.0-1-amd64 (x86_64)
Distribution : MX 18.3 Continuum
Desktop Environment : XFCE 4
-Processors-
AMD A8-9600 RADEON R7, 10 COMPUTE CORES 4C+6G : 3265,00MHz
-Display-
Resolution : 1920x1080 pixels
Vendor : The X.Org Foundation
Version : 1.19.2
-Monitors-
Monitor 0 : 1920x1080 pixels
-OpenGL-
Vendor : X.Org
Renderer : AMD CARRIZO (DRM 3.27.0, 4.19.0-1-amd64, LLVM 7.0.0)
Version : 4.5 (Compatibility Profile) Mesa 18.3.6
Direct Rendering : Yes

so...call to not existing driver ??

i tried also standard version, (master) , same output. nothing.
starting in shell, just commandline, seem working.
but no gui.

regards
Alfred

Tooniis, 2020-03-31:
This issue appeared to me when I updated mesa to 20.0.2

autotechmusic, 2020-04-25:
I got this issue after updating MESA as well.  This happens with both nightly build and stable build (Installed and AppImage).  Still haven't found a workaround.
Manjaro - KDE

AT

eduncan911, 2020-06-12:
AppImage 1.3.0

Same root cause, though I get a different error on my Thinkpad Carbon Gen 6 which uses Intel Iris GPU, no AMD like OP has.

```
$ ./Slic3r-1.3.0-x86_64.AppImage 
libGL error: MESA-LOADER: failed to open iris (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'GLXBadContext'.
  (Details: serial 1121 error_code 158 request_code 152 minor_code 6)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

Ubuntu 20.04, default kernel.

```
$ uname -a
Linux kodiak 5.4.0-7634-generic #38~1591219791~20.04~6b1c5de-Ubuntu SMP Thu Jun 4 02:56:10 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

lordofhyphens, 2020-06-12:
And the 1.3.1-dev appimage from dl.slic3r.org/dev?

autotechmusic, 2020-06-12:
> And the 1.3.1-dev appimage from dl.slic3r.org/dev?

It's not working for me on both nightly and stable (AppImage and Package Manager).

eduncan911, 2020-06-12:
> @autotechmusic And the 1.3.1-dev appimage from dl.slic3r.org/dev?

It works!  

```
Pop_OS 20.04 (Ubuntu 20.04)
5.4.0 default kernel
Regolith WM (i3wm+gaps)
```

[image]

And btw, that's awesome for the "latest" master download in that url without having to specify the git hash.  

Think I'll setup a bash/zsh alias or daily cron/systemd that:

```
wget -o /opt/Slic3r-master-latest.AppImage https://dl.slic3r.org/dev/linux/Slic3r-master-latest.AppImage
```

eduncan911, 2020-06-12:
@alfsch1 IMO, I think this can be marked as Closed after you test the latest nightly (see previous post).

lordofhyphens, 2020-06-12:
@autotechmusic https://github.com/slic3r/Slic3r/issues/4823#issuecomment-535637310 does that resolve the issue?

I suspect what I need to do is drop support for trusty and move up to Xenial. I'm putting a branch together for that.

autotechmusic, 2020-06-24:
My current workaround with the nightly AppImage:

MESA_LOADER_DRIVER_OVERRIDE=i965

Load that before executing the AppImage and it runs fine.
Hopefully that gets resolved.

Manjaro KDE 20.0.3

sdk156, 2021-11-29:
> > @lordofhyphens  And the 1.3.1-dev appimage from dl.slic3r.org/dev?
> 
> It works!
> 
> ```
> Pop_OS 20.04 (Ubuntu 20.04)
> 5.4.0 default kernel
> Regolith WM (i3wm+gaps+gnome)
> ```
> 
> [image]
> 
> And btw, that's awesome for the "latest" master download in that url without having to specify the git hash.
> 
> Think I'll setup a bash/zsh alias or daily cron/systemd that:
> 
> ```
> wget -O /opt/Slic3r-master-latest.AppImage \
>   https://dl.slic3r.org/dev/linux/Slic3r-master-latest.AppImage
> ```
> 
> I've been collecting quite a few AppImages over the years. Really keeps the system clean from packages/bloat and encapsulates everything in a single file, easy to update and copy/sync.

thank you for mentioning this, it works for me too!!

PhatHub, 2022-11-02:
Um, anyone know how to download the dev branch since dl.slic3r.org is down?
Even the repo deb package of slic3r for Ubuntu 20.04 doesn't work unless you run `export GDB_*something*=x11;slic3r` (sorry I forget the exact env variable name).
I tried creating a desktop shortcut (and adding the export to the Exec line), but it didn't work (and the desktop shortcut stopped working correctly anyway).

kzembower, 2023-01-26:
Also got the error below when downloading and running the AppImage:
```
$ sudo mv ~/Downloads/Slic3r-1.3.0-x86_64.AppImage .
$ sudo chmod +x Slic3r-1.3.0-x86_64.AppImage 
$ sudo ln -s Slic3r-1.3.0-x86_64.AppImage slic3r
$ slic3r 
Gtk-Message: 12:09:07.001: Failed to load module "canberra-gtk-module"
libGL error: MESA-LOADER: failed to open nouveau: /usr/lib/dri/nouveau_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: nouveau
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 1437 error_code 2 request_code 149 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
$ 
```
My system is:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy
$ 
```
What does it mean on the main download pages when it says: "Linux users: you need to install the freeglut package provided by your distro" No package freeglut was found in Ubuntu 22.04

Thanks for any guidance and advice.

-Kevin