slic3r/Slic3r #4949: Preview tab crash on large models

MortarArt, 2020-03-30:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
Across multiple versions of Slic3r, I've had consistent crashes when switching to the preview tab.

**To Reproduce**
1. Add a large model (or a small model with lots of support material)
2. Click on the Preview tab (with generation enabled or completed prior)
3. Unhandled exception reported, with the option to ignore
4. A portion of the slice is visible, with the top of layer slider corresponding to where slicing stopped working.

**3D Model and Slic3r Configuration Export**
Large models.
https://justpaste.it/3up2v

**Expected behavior**
Full preview to load.

**Screenshots**
I can provide one on request.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version [e.g. 1.3.1 Dev] but occurs on various versions

**Additional context**
It's occurred consistent on two graphics cards. I have 32gb of RAM, and a beefy CPU.

foreachthing, 2020-03-30:
What's a large model? A 20 cm cube?
Do you have a specific model PS crashes on?

MortarArt, 2020-03-30:
Slicing both the files at once in this example; https://www.isinnova.it/easy-covid19-eng/

So not particularly large.

foreachthing, 2020-03-31:
Nope! No crash:
[image]

System-Info: https://pastebin.com/jj1yQgWB
Dev Build: version_2.2.0-81-g70baa0d24

MortarArt, 2020-03-31:
Yes, it's not model specific. Was that using my config?

I'm assuming it's a complicated system issue.

foreachthing, 2020-03-31:
No, it was not your config. Because I need to copy, paste and save a config file.
It would be easier if you could export a 3mf file and attach that (zipped).

MortarArt, 2020-03-31:
I don't really understand what you mean. Do you want a 3mf file of the models in the position I used, because their orientation is kind of irrelevant. I only gave the models as an example of files where they cause my preview window to crash. What I meant by config was the slicing settings.

lordofhyphens, 2020-03-31:
@MortarArt attach, in an archive of some readable form, everything someone may need to reproduce your problem.

This usually includes a sample 3d model and a ctrl-E export at minimum.

MortarArt, 2020-04-24:
Please allow me to repeat. The model doesn't matter. Only the amount of slicing paths in the preview view. I have provided the settings that I use. I suspect it's a system compatibility issue. Something wrong with the folder structure, or graphics card.

I provided a link to an example 3D model that causes this failure. As I mentioned, this has been a consistent issue with my system of years, across many versions, models and slicing settings.

MortarArt, 2020-04-24:
What more do I need to do? Produce a screencap video?

lordofhyphens, 2020-04-24:
@MortarArt what I was looking for is a concise example setup that crashes reliably for you.

Model+slicing setting for that model. I want to be sure I and @foreachthing  are testing under the same conditions that you are (as close as we can anyway).

MortarArt, 2020-04-24:
3Dbenchy.stl scaled to 250%, arranged to center of bed, using these settings; https://justpaste.it/345a4 created an unhandled exception when going to the preview tab, and failed to show the full preview, after hitting ignore. Scaling to 200% does not cause the same problem.