slic3r/Slic3r #4950: Not able to run latest 1.3.1 dev build on OSX Yosemite

Anycubic, 2020-03-31:
Hello, I'm trying to run dev releases (tried slic3r-1.3.1-dev-66af505-osx.dmg and slic3r-1.3.1-dev-2ef957d-osx.dmg) but I get this error:

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @executable_path/dylibs/libperl.dylib
  Referenced from: /Applications/Slic3r.app/Contents/MacOS/perl-local
  Reason: no suitable image found.  Did find:
	/Applications/Slic3r.app/Contents/MacOS/dylibs/libperl.dylib: code signature invalid for '/Applications/Slic3r.app/Contents/MacOS/dylibs/libperl.dylib'

How can I fix this?
Thanks a lot

lordofhyphens, 2020-03-31:
Means I need to go figure out Apple's system for code signing again.

lordofhyphens, 2020-03-31:
Unfortunately I don't have a Yosemite test system to work with.

Anycubic, 2020-04-01:
> Means I need to go figure out Apple's system for code signing again.

Actually 1.3.0 works fine, hope it can help, thanks!

th0mpy, 2020-04-06:
One thing to note, Yosemite isn't supported anymore from Apple. Which means, you might not be able to run the latest version of Slic3r. Can you update to a later version of MacOS?

lordofhyphens, 2020-04-06:
Also check one of the builds in the Jenkins folder

Anycubic, 2020-04-08:
> One thing to note, Yosemite isn't supported anymore from Apple. Which means, you might not be able to run the latest version of Slic3r. Can you update to a later version of MacOS?

I know it is not supported anymore, but I have a lot of legacy apps, my system is working very fine as is, upgrading to latest OS X will be a mess and I will do it when really required.

th0mpy, 2020-04-08:
IMHO, then you need to understand that the latest versions of software may not function with such an old version of MacOS. I think the only option, difficult or not, is to update MacOS in order to get the latest version of Slic3r.

lordofhyphens, 2020-04-08:
Builds from  https://dl.slic3r.org/dev/mac/jenkins/  still are using the
10.7 baseline.

Builds from the main folder have a minimum version of 10.12.
#close
"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

closed 2020-04-08 by lordofhyphens