slic3r/Slic3r #4952: Dev build 1.3.1 - cannot hardly see the icons in Dark Mode

th0mpy, 2020-04-05:
**Describe the bug**
Using the new build of Slic3r it is very difficult to see the non-selected icons. They are greyed out and you cannot really see them unless you click the specific icon because of MacOS' dark-mode can have an effect here. Please see the attached screenshots.

Current version of Slic3r (1.3.0) does not change due to use of Dark Mode. 

**To Reproduce**
Open application, note the icon visibility with or without MacOS dark mode enabled. 

**Expected behavior**
More contrast when using MacOS Dark mode

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS 10.15.4
 - Version 1.3.1 2ef957d

With Dark Mode enabled: [image] 

WithOUT Dark Mode enabled: [image]

lordofhyphens, 2020-06-09:
This is a wxWidgets issue. 

https://trac.wxwidgets.org/ticket/18146

th0mpy, 2020-06-09:
It appears that sxWidgets has been updated, what version does Slic3r use?

lordofhyphens, 2020-06-09:
That version is a development version (3.1.x)

Slic3r is built generally with wxWidgets 3.0, because wxPerl. 

Using 3.1 or 3.2 is going to require a bunch of patches to wxPerl as there are too many backwards-breaking changes.

3.2 is the next production version of wxWidgets.