slic3r/Slic3r #4956: Print each stl file with a different extrusor via command line

rchacon83, 2020-04-17:
Good morning,
 
I think would be very useful to develop a way to print several stl with different extrusor via command line.

I am thinking in something like this:
slic3r [ ACTION ] [ OPTIONS ] [ model1.stl --e1 model2.stl --e2 ... ]

This would produce the same result as to select a different extrusor in objet settings in gui.

I am developping multiples changes in Slic3r to apat it to a our metal 3D printer, sorry for not share the code anywhere, because I have made deep changes like how to generate the gcode, how to calculate the amount of metal extruded, etc. This changes make the software incompatible with other printers, so sorry for not sharing the code, anyway if someone is interested in see the code I could send of course.

lordofhyphens, 2020-04-17:
Regarding code sharing: remember that Slic3r is GPL licensed. You are required to abide by the terms of the licence if you distribute your modified Slic3r to anyone.

rchacon83, 2020-05-11:
Can no one help me out with this?