slic3r/Slic3r #4958: Slower than expected layers.

rewolff, 2020-04-23:
**Describe the bug**
When slicing my models I often get layers that are suddenly WAY slower than expected. It is as if it slows things down because of the minimal layer time, but on a layer that doesn't need it. 

**To Reproduce**
I'm going to try to attach the model and settings files. If github doesn't let me I'll post a link to a zipfile. 

When slicing the attached model, it does the first layer REALLY, REALLY slowly. The whole thing prints in two hours, the first layer takes over an hour. 

**3D Model and Slic3r Configuration Export**
 [smdc.zip](https://github.com/slic3r/Slic3r/files/4520428/smdc.zip)

**Expected behavior**
I expect the first layer to be printed at a reasonable speed, My setting is "50%". In this model the second layer is close to identical, so when that prints in 2:20 the first layer (excluding skirt)  should be able to print in about 4:40. In the current situation it takes over 10 times longer. 

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Linux. (Ubuntu 18.04)
-  Version 1.3.0

**Additional context**
Many other models exhibit the same problem. But usually not at the first layer. When you're checking in on the print sometimes it is going really really slow.

lordofhyphens, 2020-04-23:
Autospeed.

rewolff, 2020-04-23:
FYI: I found a link to "issues" as: "communicate with us". 

The hint "autospeed" does not help me. 
I have a bunch of speeds set to "auto" thinking that the default will be reasonable. 

Are you saying that the bug is in "autospeed"?
Are you saying that I need to turn autospeed on?
Are you saying that I should turn autospeed Off?

lordofhyphens, 2020-04-23:
Autospeed is known to do weird things here and there, especially when you have a model with a lot of gap fill. This is because it's trying to maintain constant pressure in the hotend to minimize retraction.

rewolff, 2020-04-23:
So I should turn "autospeed off" becuase it is known to do weird things, right?

I think it is already off. 
[image]

rewolff, 2020-04-23:
FYI: I created an SED script that creates an SED script that edits the GCODE to enforce a minimum speed of 1600 mm/min. This resulted in the first layer for this project to go from an hour to 7.5 minutes.  It looks like print quality suffers a bit, but I'm going to get functional objects in a reasonable time.

lordofhyphens, 2020-04-23:
Also: did you test with 1.3.1-dev?

rewolff, 2020-04-24:
OK. Now I have. Same thing. First layer: 28 minutes, OK. Without skirt: 24 minutes. The first five layers are all very similar, so I expect similar times. The first layer should be something like 160% width at 50% speed, so about 80% overall speed: 25% slower than the second and third layer. Those are estimated to take 2:13 (that goes down to 2:06 when the big solids go to 30% infill, but it is the last layer (toplayer) of the "bottom" of the object. 

Still suspecting that I've used a setting that causes this, I tried the following: remove all settings, start fresh with all defaults. Plan: look for the settings file(s) store them in a safe place for now, delete all of them and start fresh. Restore backup if no difference. I looked for documentation on where the settings are stored and the documentation told me: "You don't need to know". Fine. So I can't run that test.

lordofhyphens, 2020-04-24:
On Linux and osx all of the profile settings are under ~/.Slic3r

That should have been written down somewhere in the docs.

Thanks for checking with 1.3.1-dev.

rewolff, 2020-04-24:
FYI: I tried looking for ~/.s* but found nothing.... 

So I'm "safe" if I move ~/.Slic3r away to .slicer_bak, start a new instance, test that and then move the backup back. Right?

rewolff, 2020-04-24:
Ok. Backed up my .Slic3r directory, started Slic3r, went through the configuration wizard, and changed just a few settings: layer height: 0.2, first layer height 0.3, more top-solid-layers and infill from 20->30%. 

The first layer now claims to print in 4 minutes. So we've found that it is SOME setting that I've touched.... 

I have a "new" printer here at the office, that requires some break-in-test-prints, and an object that I don't mind having a few more of.... Lets see what happens.