slic3r/Slic3r #4961: wrong Gcode (F3.8683e+06)

hexxter, 2020-04-27:
Slic3r version 1.3.0

it generate sometime for F wrong values like "F3.8683e+06" and this is Interpreted as 3.8xx and the print stand still.

My settings for speed ist a max Value and all other as auto.

BR Hexxter

hexxter, 2020-04-27:
After an Restart of Slic3r the wrong values are gone but after a while of use it comes back.

lordofhyphens, 2020-04-27:
@hexxter and 1.3.1-dev?

hexxter, 2020-04-27:
sorry i have only the version 1.3.0 downloaded from slic3r.org

lordofhyphens, 2020-04-27:
https://dl.slic3r.org/dev

hexxter, 2020-04-27:
so far as i see in my first tests is the error fixed in version 1.3.1-dev

hexxter, 2020-04-28:
New Error in Gcode with Version 1.3.1-dev.
[applejack_mane.zip](https://github.com/slic3r/Slic3r/files/4547780/applejack_mane.zip)
Line 77414

G1 F8372295714209792.000

i think this is not a right gcode ?

lordofhyphens, 2020-04-29:
Please use ctrl-E export, not bundles, for configurations.

lordofhyphens, 2020-04-29:
````
gap_fill_speed = 0
````

Looks like you are using autospeed, which has some really weird edge cases when gap fill is involved.

Please set manual speeds instead of autospeed and see if it is still giving really high speeds.

hexxter, 2020-04-29:
ok i think its not easy to make auto speed in every case but i think it should be a max speed set when crasy values are calculated? And set the max value?

hexxter, 2020-04-29:
set of gap_fill_speed to a value has not fixed it. I have set all speed now the problem is gone... But i think this is not a good solution?