slic3r/Slic3r #4965: Saving printer settings causes slic3r to crash

mechsin, 2020-05-02:
**Issue**
I am new to slic3r and I was trying to setup my printer but it every time I save my printer settings slic3r crashes. After this it also fails to start up again giving off a free: invalid pointer message. This happens even if I don't make any changes. Slic3r will start again if I remove the ini file that was created when the save occurred. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open printer settings 
2. Hit the button to save current printer settings
3. Enter save name, I have just been leaving the default Untitled but I have also use Anet
4. Hit OK and slic3r crashes
5. Slic3r crashes silently on start until Untitled.ini file is removed, in my case from ~/.Slic3r/printer

**3D Model and Slic3r Configuration Export**
No model 

**Expected behavior**
Slic3r shouldn't crash and shouldn't fail to start. If there is something wrong with the printer configuration the user should get a warning prior to the file saving.

**Desktop (please complete the following information):**
 - OS: Linux, Manjaro
 - Version: 1.3.0

**Additional context**
This is the ini file that is created on save
[Untitled.txt](https://github.com/slic3r/Slic3r/files/4569249/Untitled.txt)

lordofhyphens, 2020-06-09:
Check with 1.3.1-dev

mechsin, 2020-07-17:
I was able to use the Slic3r-1.3.1-dev-877138e-PR4843-x86_64.AppImage of the dev build and everything seemed to work fine no crashes. I did the test as I stated in the original post as well as editing some of the settings and saving the values as a new printer. I was able to both save the settings and close and restart Sli3r with crashing.

So it looks like this issue will be fixed when you release 1.3.1

Thanks

TheAlgorythm, 2020-11-17:
I have the same error on Manjaro but on slicing.

```sh
$ slic3r
free(): invalid pointer
[1]    54455 abort (core dumped)  slic3r
```

The first time I opened slic3r all was ok until i tried to slice an amf-file. Then slic3r crashed and since then it is not possible to start slic3r anymore.
The only thing which won't crash is slicing from the CLI.
I'm using the AUR binary, which downloads it directly from the GitHub releases page. I'm currently on v1.3.0-3.

x14km3d, 2021-09-22:
DISTRIB_ID=ManjaroLinux
DISTRIB_RELEASE=21.1.3
DISTRIB_CODENAME=Pahvo
DISTRIB_DESCRIPTION="Manjaro Linux

Since it seems that the problem still could not be solved, I'm warming up the topic again. I have downloaded the latest version from the project page.  I can fill out the welcome assistant completely, but when I import an stl file and change the print settings afterwards, the software crashes when I press the save button. After that I can't start the software because it also ends with an invalid pointer message.

alexban011, 2022-05-31:
also happens on archlinux 5.18.1-arch1-1