slic3r/Slic3r #4967: Cannot compile when using Boost 1.73

fluffysheap, 2020-05-11:
### Version
git head, version 54a31eed204863c49dfaa3b6fe4809f6d10a94ba

### Operating system type + version
Gentoo Linux unstable (rolling release date: May 9)

### Behavior
Cannot compile when using Boost 1.73.  Boost now requires bind.hpp to be included explicitly when using placeholder arguments.  Typical error resembles the following: 
src/libslic3r/PrintObject.cpp: In member function ‘void Slic3r::PrintObject::infill()’:
src/libslic3r/PrintObject.cpp:1214:49: error: ‘_1’ was not declared in this scope
 1214 |         boost::bind(&Slic3r::Layer::make_fills, _1),

The source files that need to be updated are:
src/libslic3r/SLAPrint.cpp
src/libslic3r/PrintObject.cpp
src/libslic3r/SupportMaterial.cpp
src/libslic3r/TriangleMesh.cpp

#### STL/Config (.ZIP) where problem occurs
No STL or config is relevant, but I attached a patch file that fixes the problem.

[boost-headers.patch.gz](https://github.com/slic3r/Slic3r/files/4607849/boost-headers.patch.gz)

zueski, 2020-05-23:
Came here to post the same issue from Gentoo.  To build, I also needed to comment `-Wno-undefined-var-template` from xs/Build.PL as gcc was complaining about an removed option.  The patch worked fine for me.

hroncok, 2020-06-02:
We have 3 patches in Fedora to build it. Will open a PR here.

jwakely, 2020-06-02:
https://src.fedoraproject.org/rpms/slic3r/blob/936cb29b1a703310cceb51e4c77993c27c8dae75/f/slic3r-bind-placeholders.patch is necessary due to changes in Boost.Bind.

https://src.fedoraproject.org/rpms/slic3r/blob/936cb29b1a703310cceb51e4c77993c27c8dae75/f/slic3r-boost-nowide.patch is necessary to use the version of Boost.Nowide included in Boost 1.73.0 (possibly not interesting to you because you bundle an older version of nowide, but we don't use that in Fedora).

https://src.fedoraproject.org/rpms/slic3r/blob/936cb29b1a703310cceb51e4c77993c27c8dae75/f/slic3r-endian.patch is necessary because the `<boost/detail/endian.hpp>` header and  `BOOST_LITTLE_ENDIAN` macro are no longer provided by Boost. The solution there uses [GCC macros](https://gcc.gnu.org/onlinedocs/cpp/Common-Predefined-Macros.html), which are also defined by Clang, but you might want a more portable solution.

hroncok, 2020-06-02:
> slic3r-endian.patch is necessary because the <boost/detail/endian.hpp> header and BOOST_LITTLE_ENDIAN macro are no longer provided by Boost. The solution there uses GCC macros, which are also defined by Clang, but you might want a more portable solution.

Fortunately, this has been already dropped in master: ce966c2450183dfc1c6415161463ea8d59dcccd5

The remaining 2 patches are in https://github.com/slic3r/Slic3r/pull/4976

lordofhyphens, 2020-07-23:
@fluffysheap can you check branch slic3r/boost173 ?

VanessaE, 2020-12-16:
This issue also holds true for boost 1.74 - can't build against that version either.  Same error.

VanessaE, 2021-01-08:
I didn't notice the `boost173` branch comment before.  That also fails to build, with the same error (though on a different line).

PR #4996 (2021-03-14): Backwards-compatible Boost 1.73
Amend work submitted by @hroncok to allow for building on Boost 1.73 while maintaining compatibility for older versions.

Fixes #4967
closed 2021-03-14 by lordofhyphens