slic3r/Slic3r #4968: Support material interface: invalid speed and zero extrusion

centripetal3, 2020-05-11:
### Version
1.3.1-dev c2aa7e7

### Operating system type + version
Ubuntu 18.04 LTS

### Behavior
When generating support material interface, Slic3r will sometimes produce a zero extrusion and an invalid print speed. Example G-code below:

; layer is 30
G1 X96.635 Y108.936 F12000.000 ; move to first support material interface point
G1 Z2.400 F12000.000 ; restore layer Z
G1 E2.00000 F1500.00000 ; unretract extruder 0
G1 F26310536999.081   ; <---- NOTE: Invalid speed
G1 X96.618 Y108.878 E0.00000 ; support material interface  <---- NOTE: Zero extrusion

When the printer runs this code, it stalls indefinitely.

#### STL/Config (.ZIP) where problem occurs
[support-material-test-case.zip](https://github.com/slic3r/Slic3r/files/4611034/support-material-test-case.zip)

Any idea what's happening here? ...anything I can do to help diagnose this further?

supermerill, 2020-05-11:
ok, it makes sense, you can print a zero-length extrusion at the speed of light!

centripetal3, 2020-05-11:
I just noticed this may be related to #4961
https://github.com/slic3r/Slic3r/issues/4961