slic3r/Slic3r #4971: Slic3r adding defective additional filling layer inside the object

FPrefect20, 2020-05-19:
**Describe the bug**
When slicing the enclosed cone object an 0.2mm filling layer is added inside the object and the object is closed with 1 layer filament.

**To Reproduce**
1. Open the enclosed object with slic3r
2. Rotate with "3d view" to have a look inside (there's no filling inside)
3. Click on "preview"
4. After the slicing completes the additional layer can be observed inside the object

**3D Model and Slic3r Configuration Export**
[slic3r cone.zip](https://github.com/slic3r/Slic3r/files/4652229/slic3r.cone.zip)

**Expected behavior**
No additional filling added

**Screenshots**
[image]

Printing:
[image]

**Desktop (please complete the following information):**
 - Windows 10  64Bit
 - Version 1.3.0

lordofhyphens, 2020-05-19:
https://dl.slic3r.org/dev/win can you reproduce with 1.3.1-dev?

FPrefect20, 2020-05-20:
Yes, same problem.

[image]

lordofhyphens, 2020-05-20:
Thanks. One more sanity check: did you pass the model through https://service.netfabb.com/service.php

FPrefect20, 2020-05-20:
Although other slicer software doesn't show the "additional filling layer" problem the fixing with "netfabb" indeed seems to solve the problem. In the "preview" of Slic3r the object is shown correct now.
I will print the object again to double check the result.
Thank you!!!

closed 2020-06-09 by lordofhyphens