slic3r/Slic3r #4972: Expose overlap_factor

rbn920, 2020-05-24:
In the section of the manual about flow math it discusses `overlap_factor`. I may be wrong but I believe this could be a useful variable to expose to the user. Particularly, I believe this could solve my issue with solid layers when using larger (ie 0.8mm) nozzles. Although, perhaps it is not necessary because the this can sort of be handled by the extrusion width settings.

lordofhyphens, 2020-05-24:
I believe the overlap factor is currently a constant. To make it user configurable you would need to pass it into the flow math functions it is used in, in addition to adding it to the gui/config stack

rbn920, 2020-05-25:
I see. I was hoping it could be a simpler task. I'll go ahead and close the issue. I think a more useful request pertaining to my particular printing issue would be extrusion multiplier overrides for patrticular features (ie perimeters vs solid layers).

closed 2020-05-25 by rbn920