slic3r/Slic3r #4973: z lift - option to lift during travel, instead of before travel.

paulnewall, 2020-05-25:
**Problem is:**
petg oozes, causing strings, and blobs at the end of travel moves when the ooze gets wiped off on the perimeter. Z lift can reduce the blobs by carrying them over the perimeter at the end of the travel, but z lift at the start of the travel makes strings worse by lifting slowly off the surface.
**Proposal:**
do the z lift at the same time as the travel (xyz move together) then the departure from the surface is fast, reducing strings. At the end of the travel z has lifted, reducing blobs. Unlift is still a vertical downward move.
**Implementation:**
I modified libslic3r in prusaslicer to implement this and it is working very well. It's not a trivial change because the existing code has the retract and lift code rather mixed together. I imagine making this an option would be safer than changing the existing behaviour of z lift in case some users want to keep the vertical z lift. Screen shot shows a horizontal view of the blue travel moves.
[image]

lordofhyphens, 2020-05-25:
And of course it won't help firmware retraction.

lordofhyphens, 2020-05-25:
Pull request with a test will go further :)