slic3r/Slic3r #4984: generated g-code has negative extruder direction

jin-eld, 2020-06-14:
Hi,

I tried out slic3r recently and right from the start I ran into a strange problem. At fist I thought that I may have applied some weird setting somewhere, but after going through all checkboxes and options I could find out what it may be... so perhaps its a bug after all.

**Describe the bug**

Generated g-code has negative extrusion direction.

I loaded an STL (export from OnShape), sliced it and observed that when the print starts, extruder is running backwards. It's not a reversed pin setting on the printer, in my "start g-code" I have a nozzle cleaning line which prints correctly.

I then looked at the generate g-code and indeed E values are negative, here's a snippet:
`G1 X54.981 Y99.934 E-1.64565
G1 X165.019 Y99.934 E-4.44879
G1 X165.066 Y99.981 E-4.45048
G1 X165.066 Y120.019 E-4.96095
G1 X165.019 Y120.066 E-4.96264
G1 X54.981 Y120.066 E-7.76578
G1 X54.934 Y120.019 E-7.76747
G1 X54.934 Y100.017 E-8.27701
`

**To Reproduce**

Steps to reproduce the behavior:
1. Load the STL (attached in zip)
2. Apply settings (configuration settings attached in zip)
3. Go to "Quick Slice and Save As..." and save the gcode.
4. Inspect the resulting g-code (attached in zip)

**Expected behavior**

Positive extrusion direction...

**Desktop (please complete the following information):**
 - OS: Linux, Fedora 31
 - Version: 1.3.0, also tried master appimage 1.3.1-dev  / Build commit: 6eeea77

[issue.zip](https://github.com/slic3r/Slic3r/files/4777358/issue.zip)

dwillmore, 2020-06-15:
Thanks for the bug report.  I can't say specifically what is the cause of the negative extrusion direction, but I can tell you what setting is causing it.  If you look at your Print Settings->Advanced->Extrusion Width->First Layer setting.  You have it as auto.  I'm not sure what width it's picking, but I'm guessing it's negative.  You have some odd values set.  The first layer height of 0.8mm is way too large for your 0.4mm nozzle.  The maximum you would want to use with a 0.4mm nozzle is 80% of that or 0.32mm.  Also, your widths are very narrow.  Typical widths are 100% to 150% of your nozzle width.  You have values of 0.1 and 0.2mm.  You will get very unreliable extrusion with those settings.  

If I change the first layer width to 0.2mm (to match your other values), then I get valid positive extrusion lengths.  I doubt one would get a successful print, but it's better looking gcode.

I would suggest upping your first layer width to 0.6mm and your perimeters to 0.4mm and your external perimeters to 66%.  That's about as narrow as you would want to go with an 0.4mm nozzle--unless you've done a lot of work to verify that your printer can handle narrower draws.

Problem reproduced with 1.3.1 build dev-adea0d9-x86_64 AppImage.

dwillmore, 2020-06-15:
Try this one.
[try.zip](https://github.com/slic3r/Slic3r/files/4781151/try.zip)

jin-eld, 2020-06-15:
@dwillmore Thank you very much! Also for the help on IRC, I think with your help we now know which of my settings was wrong. I had "Perimeters" set to a certain width in mm, but I left "Perimeters -> external" at the default setting, which I think was 66% and this of course did not work out. Once I started controlling the external perimeter in mm, things became clearer.

That being said, it would still be great to have a fix for the negative extrusion values bug or at least some sanity checker which would warn about incorrectly set values.